raspberrypi/userland #212: Feature Request: Single Picture, No Timeout

ad0nis, 2015-01-30:
It appears to be impossible to get raspistill to take a single picture without a delay. One would think that setting a timeout of 0 and no preview, as in the following example would accomplish this:

$ raspistill -n -t 0 -o test.jpg -v

raspistill Camera App v1.3.8

Width 2592, Height 1944, quality 85, filename test.jpg
Time delay 0, Raw no
Thumbnail enabled Yes, width 64, height 48, quality 35
Link to latest frame enabled  no
Full resolution preview No
Capture method : Run forever

Preview No, Full screen Yes
Preview window 0,0,1024,768
Opacity 255
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
Camera component done
Encoder component done
Starting component connection stage
Connecting camera preview port to video render.
Connecting camera stills port to encoder input port
Opening output file test.jpg
Enabling encoder output port
Starting capture 1
Finished capture 1
Opening output file test.jpg
Enabling encoder output port
Starting capture 2
Finished capture 2
Opening output file test.jpg
Enabling encoder output port
Starting capture 3
Finished capture 3
^Cmmal: Aborting program

I believe the problem is here:
https://github.com/raspberrypi/userland/blob/ffcc4bd7c4875b71376c4240116e251652c9bec0/host_applications/linux/apps/raspicam/RaspiStill.c#L491-492

In this situation, I neither want to run forever, nor do I want to have to press a key to take the picture. I want an image created directly after the command is executed. Is there some option that I've missed?

JamesH65, 2015-01-30:
It's not possible to take the picture instantly - the camera needs to 
start up, start streaming, get its exposure right, stop streaming and go 
in to stills mode, take the picture, go back to streaming, shut down. 
That takes about 500ms. So try a shorter and shorter timeouts until the 
picture fails to take (may be black if exposure incorrect).

6by9, 2015-01-30:
It would be possible to do this, but ONLY if the shutter speed and ISO are manually set on the command line. Otherwise (as James says) the AGC and AWB algorithms have no chance to converge on sensible values. If you are prepared to accept that then the issue can be investigated further, but otherwise the recommendation would be to allow at least 500ms.

(The AWB convergence can be circumvented by adding -st to the command line to force the GPU to compute AWB values based on the actual capture frame, however this slows down the capture process).

ad0nis, 2015-01-31:
I am sorry if my initial use of the phrases 'directly after' and 'without a delay' threw the conversation off the point that I was really trying to get at. Could there just be a CLI option that does all the measurements and calculations that you're talking about, and then returns as quickly as possible, instead of having to wait 5 seconds? I am just talking about simplicity and usability. If I may suggest '-q' as the new option, as a simple mnemonic for quick? (or '-qs' for quick snapshot?)

6by9, 2015-02-01:
Er, "-t 500 -st"? 
-st as I mentioned means AWB calcs are done on the actual capture frame stats rather than based on preview. -t 500 gives AGC a chance to have adapted (it won't be perfect in that time, but pretty reasonable).

Ruffio, 2015-06-27:
@ad0nis is this still an issue?

ad0nis, 2015-10-01:
I've put the project I was going to use this tool for on hold for now, so it's not an issue for me personally at the moment, but I can imagine that other users would still appreciate a convenience function as described earlier.

JamesH65, 2017-12-05:
As far as I can tell, everything required to do this can be done already, so closing. Its also pretty old...

closed 2017-12-05 by JamesH65

defic, 2018-06-19:
I am using python trying to cut the delay to minimal and using solution posted by Entryton ( https://raspberrypi.stackexchange.com/a/75464 )

However, the delay is actually as big as 500ms with this method, and not 50ms as Entryton claims. Is there something that can be improved with the script? Other option I had in mind was a capturing image stream to a buffer and capturing from it...

For my project, I need to take a picture as soon as a sensor is triggered. Thanks in advance!

JamesH65, 2018-06-19:
Getting a decent image in 50ms  doesn't sound particularly possible.  Have you read the entirety of this issue?

What resolution do you need to capture? You could use video recording, dispose of unneeded frames, and then keep the frame at the point the sensor is triggered.

dustinkerstein, 2020-11-06:
@defic did you ever find a solution for this? I'm running into the same thing when testing that server/client design in the link you posted. I've gotten the latency down to around 200-300ms when using burst mode + continuous capture of picamera, but I really need to get <100ms, and with very little variation between shots.