raspberrypi/userland #213: Possible missing header in libraspberrypi-dev package?

jusme101, 2015-02-05:
Building a custom mmal component based on passthrough.c from git://github.com/raspberrypi/userland.git needs mmal_events_private.h, which is not installed in /opt/vc/include/interface/mmal/core/ from the libraspberrypi-dev package, although it's #included from mmal_port_private.h, which is present.
                                                                  
By copying the file from a git clone all is well, but I'd like the code to be buildable just using the installed packages.

asb, 2015-02-05:
Thanks for the report. The libraspberrypi-dev package is built from the firmware repo, which doesn't contain this header file. Sounds like a bug for @popcornmix

popcornmix, 2015-02-05:
Yes, will add in next firmware update.

jusme101, 2015-02-06:
Many thanks!

jvcleave, 2015-03-08:
Any chance of getting this in a gist?

popcornmix, 2015-03-08:
This was added a month ago:
https://github.com/raspberrypi/firmware/blob/master/opt/vc/include/interface/mmal/core/mmal_events_private.h
https://github.com/Hexxeh/rpi-firmware/blob/master/vc/sdk/opt/vc/include/interface/mmal/core/mmal_events_private.h

closed 2015-03-08 by popcornmix