raspberrypi/userland #214: Broken support for RPI 2?

agherzan, 2015-02-05:
For whatever reason I can't compile userland for rpi 2 (TUNE_FEATURES     = "arm armv7a vfp neon" TARGET_FPU = "vfp-neon"):
libvchostif.a: error adding symbols: Bad value

Is there a known broken support for RPI 2?

unphased, 2015-02-05:
TIL about RPi 2. Thanks, @agherzan and GitHub. 

I have zero authority on this matter but i think that, considering the whole new hardware redesign, it's perhaps just a little ludicrous to expect this to just work out-of-box. 

Hopefully it won't take too long though, since it's supposed to be fully compatible. Which is friggin' sweet, too.

agherzan, 2015-02-05:
The only change is the BCM soc. So it shouldn't be much more as they already have the binaries in the firmware AFAIK

popcornmix, 2015-02-06:
You are expected to build userland just how you always have.
Raspbian and the userland libs are designed to run on both pi1 and pi2 so build with pi1 compile flags.
If you want to build explicitly for pi2 (and break compatibility with pi1) you are free to do so, but it's not something directly supported by us.
Note userland is a fairly thin library for communicating with the gpu. I don't belive you'll see any performance benefit from pi2 specific compile flags.

agherzan, 2015-02-06:
Wait a sec. The soc was changed. You have to use different compiler flags and so on. I didn't change anything else than using the tool-chain and flags and so on for the cortexa9 architecture.

popcornmix, 2015-02-06:
You don't have to use different compiler flags. The armv7 is compatible with armv6 code. On Pi 2
```
git clone https://github.com/raspberrypi/userland.git
cd userland
./buildme
```
Completed just fine. (You may need to apt-get install cmake)

agherzan, 2015-02-06:
Well I'm using yocto as I maintain the BSP layer for that BSP. And in the process of supporting RPI2 I updated the machine configuration which defines the armv7 CPU to take advantage of neon, VFP4 and so on. I understand and I know that I can run arm6 code on arm7 but this doesn't mean it is efficient.

pelwell, 2015-02-06:
If you are compiling userland with non-standard flags then I'm not surprised if it doesn't "just work". However, if you do get it working then feel free to submit a patch.

agherzan, 2015-02-06:
So let me understand. You know you have a more powerful engine under the hood and you just want to stick with this workaround? Just to make sure we are on the same page. And there is no plan of getting this right in the near future, right?

Ferroin, 2015-02-06:
On 2015-02-06 11:40, Andrei Gherzan wrote:
> So let me understand. You know you have a more powerful engine under the
> hood and you just want to stick with this workaround? Just to make sure
> we are on the same page. And there is no plan of getting this right in
> the near future, right?
No, things are working right, they have explicitly said multiple times 
that they don't intend to break compatibility with the original RPi.  If 
you actually look at the code, you will see that it really is (almost) 
all just minimalistic wrappers around hardware level commands, compiling 
for armv7 isn't going to get you more than maybe a few nanoseconds in 
performance difference tops, and probably won't provide any better 
performance at all.  The only thing that would likely improve in 
performance is vcfiled, and that is still not very likely, as it spends 
most of it's time sleeping.

agherzan, 2015-02-06:
I understand. And this is perfectly valid as long as we are talking only from the perspective of userland. But think of people who want to compile stuff with neon support for example. They will need to use a custom toolchain which won't be able to be used in compiling userland. The only option they have is to use the binaries in firmware. That's a very valid use-case which is broken right now.

pelwell, 2015-02-06:
Future plans are future plans. Right now, we won't be making any changes in that area.

agherzan, 2015-02-06:
Fair enough. Please don't close this issue.

popcornmix, 2015-02-06:
But I don't see what is broken.

Are you trying to build on pi or cross-compile?
The cross-compile toolchain supports -mcpu=cortex-a7 -mfpu=neon-vfpv4
https://github.com/raspberrypi/tools/tree/master/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64

gcc-4.8 on Pi also supports those flags.
For default gcc (4.6) you need -march=armv7-a -mfpu=neon-vfpv4

See this thread: http://www.raspberrypi.org/forums/viewtopic.php?f=33&t=98354

agherzan, 2015-02-06:
I'm using gcc 4.9.1 with the following CFLAGS:
export CC="arm-poky-linux-gnueabi-gcc  -march=armv7-a -mfloat-abi=softfp -mfpu=neon --sysroot=/tmp/sysroots/raspberrypi2"

JamesH65, 2015-02-06:
I've been using the standard cross compiler (Linaro) with the extra flags stated above, for compiling x264 with cortex-a7/NEON support. I also use the same tool chain to build userland, with the standard build.

So the same toolchain works for both, just with different flags. And of course the binaries coexist fine.

agherzan, 2015-02-06:
That's interesting. So there must be an issue somewhere else... Let me come after a retest the compiler.

agherzan, 2015-02-06:
Question. What happens if I mix on RPI2 armv6 with armv7? Example library armv6 (gl stuff) and an app armv7.

JamesH65, 2015-02-06:
Not sure, but I *think* that I was using a Armv6 build of avconv with a cortex-a7/NEON libx264 today, and that worked fine. I'm not sure though! I don't see why they wouldn't work - the calling convention is the same, just the instructions used in the library/app may differ.

agherzan, 2015-02-06:
I tried to compile the entire platform as cortex-a7/NEON and use the binaries for GL libraries. And for whatever reason omxplayer is not able to use th gl libraries:
/usr/bin/omxplayer.bin: error while loading shared libraries: /usr/lib/libWFC.so: internal error

And that confused me a little thinking that must be an incompatibility that I don't figure out. Anyway. I will close this for now and come back when I have some more info.

Thanks all for support!

agherzan, 2015-02-08:
Ok. So I managed to point the issue and find a workaround in my current environment. The complete log is:

---
libvchostif.a(vc_vchi_gencmd.c.o): relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
libvchostif.a: error adding symbols: Bad value
---

Checking the compile flags I see that most of the objects are compiled with fPIC while vc_vchi_gencmd is not. So, I managed to compile everything just fine by forcing a -fPIC in CFLAGS.

agherzan, 2015-02-08:
Shouldn't "add_library(vchostif" be SHARED?

popcornmix, 2015-02-08:
Can you explain exactly what you changed?
Did you edit makefiles/cmake/toolchains/arm-linux-gnueabihf.cmake (or make a copy of it)?
What if the contents of your .cmake file?

agherzan, 2015-02-08:
Below is the entire cmake file generated by the buildsystem (I think I mentioned above that I'm using this in yocto buildsystem):

```
set( CMAKE_SYSTEM_NAME Linux )
set( CMAKE_SYSTEM_PROCESSOR arm )
set( CMAKE_C_COMPILER arm-poky-linux-gnueabi-gcc )
set( CMAKE_CXX_COMPILER arm-poky-linux-gnueabi-g++ )
set( CMAKE_ASM_COMPILER arm-poky-linux-gnueabi-gcc )
set( CMAKE_AR arm-poky-linux-gnueabi-ar CACHE FILEPATH "Archiver" )
set( CMAKE_C_FLAGS " -march=armv7-a -marm  -mthumb-interwork -mfloat-abi=hard -mfpu=neon-vfpv4 -mtune=cortex-a7  --sysroot=/media/Personal/yocto/BUILD-master/tmp/sysroots/raspberrypi2  -O2 -pipe -g -feliminate-unused-debug-types" CACHE STRING "CFLAGS" )
set( CMAKE_CXX_FLAGS " -march=armv7-a -marm  -mthumb-interwork -mfloat-abi=hard -mfpu=neon-vfpv4 -mtune=cortex-a7  --sysroot=/media/Personal/yocto/BUILD-master/tmp/sysroots/raspberrypi2  -O2 -pipe -g -feliminate-unused-debug-types -fvisibility-inlines-hidden" CACHE STRING "CXXFLAGS" )
set( CMAKE_ASM_FLAGS " -march=armv7-a -marm  -mthumb-interwork -mfloat-abi=hard -mfpu=neon-vfpv4 -mtune=cortex-a7  --sysroot=/media/Personal/yocto/BUILD-master/tmp/sysroots/raspberrypi2  -O2 -pipe -g -feliminate-unused-debug-types" CACHE STRING "ASM FLAGS" )
set( CMAKE_C_FLAGS_RELEASE "-O2 -pipe -g -feliminate-unused-debug-types  -O2 -pipe -g -feliminate-unused-debug-types -DNDEBUG" CACHE STRING "CFLAGS for release" )
set( CMAKE_CXX_FLAGS_RELEASE "-O2 -pipe -g -feliminate-unused-debug-types  -O2 -pipe -g -feliminate-unused-debug-types -fvisibility-inlines-hidden -DNDEBUG" CACHE STRING "CXXFLAGS for release" )
set( CMAKE_ASM_FLAGS_RELEASE "-O2 -pipe -g -feliminate-unused-debug-types  -O2 -pipe -g -feliminate-unused-debug-types -DNDEBUG" CACHE STRING "ASM FLAGS for release" )
set( CMAKE_C_LINK_FLAGS " -march=armv7-a -marm  -mthumb-interwork -mfloat-abi=hard -mfpu=neon-vfpv4 -mtune=cortex-a7  --sysroot=/media/Personal/yocto/BUILD-master/tmp/sysroots/raspberrypi2  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed" CACHE STRING "LDFLAGS" )
set( CMAKE_CXX_LINK_FLAGS " -march=armv7-a -marm  -mthumb-interwork -mfloat-abi=hard -mfpu=neon-vfpv4 -mtune=cortex-a7  --sysroot=/media/Personal/yocto/BUILD-master/tmp/sysroots/raspberrypi2  -O2 -pipe -g -feliminate-unused-debug-types -fvisibility-inlines-hidden -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed" CACHE STRING "LDFLAGS" )

# only search in the paths provided so cmake doesnt pick
# up libraries and tools from the native build machine
set( CMAKE_FIND_ROOT_PATH /media/Personal/yocto/BUILD-master/tmp/sysroots/raspberrypi2 /media/Personal/yocto/BUILD-master/tmp/sysroots/x86_64-linux    )
set( CMAKE_FIND_ROOT_PATH_MODE_PACKAGE ONLY )
set( CMAKE_FIND_ROOT_PATH_MODE_PROGRAM ONLY )
set( CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY )
set( CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY )

# Use qt.conf settings
set( ENV{QT_CONF_PATH} /media/Personal/yocto/BUILD-master/tmp/work/raspberrypi2-poky-linux-gnueabi/userland/git-r4/qt.conf )

# We need to set the rpath to the correct directory as cmake does not provide any
# directory as rpath by default
set( CMAKE_INSTALL_RPATH  )

# Use native cmake modules
set( CMAKE_MODULE_PATH /media/Personal/yocto/BUILD-master/tmp/sysroots/raspberrypi2/usr/share/cmake/Modules/ )

# add for non /usr/lib libdir, e.g. /usr/lib64
set( CMAKE_LIBRARY_PATH /usr/lib /lib)
```

popcornmix, 2015-02-11:
It seems to be an issue that shared libraries are implicitly built with -fPIC and static libraries without.
For some reason an armv6 build is happy mixing the two, but an armv7 build complains.

While in theory we could build all libs as shared, that does have repercussions (applications like xbmc that ship with some libraries from /opt/vc/lib will now need to include the extra shared libs, like vchostif).

I'm planning to add this patch
```
commit 5fe4cd035c684049928407dbf640cda3cb2cd03e
Author: popcornmix <popcornmix@gmail.com>
Date:   Wed Feb 11 14:35:11 2015 +0000

    cmake: Add -fPIC by default to avoid static/shared linking issues with armv7

diff --git a/CMakeLists.txt b/CMakeLists.txt
index ee6edd6..d8f776c 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -30,7 +30,7 @@ include_directories(host_support/include)
 
 # Global compiler flags
 if(CMAKE_COMPILER_IS_GNUCC)
-   set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wno-multichar -Wall -Wno-unused-but-set-variable")
+   set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wno-multichar -Wall -Wno-unused-but-set-variable -fPIC")
 endif()
 
 add_definitions(-D_REENTRANT)
```
which should give that same effect as you discovered from adding -fPIC to CFLAGS.

popcornmix, 2015-02-13:
userland updated with proposed patch: https://github.com/raspberrypi/userland/commit/c5341f09dc686c17966369485f7dd27f58dc081c
@agherzan okay with you?

agherzan, 2015-02-13:
Brilliant. Thanks @popcornmix .

closed 2015-02-13 by agherzan