raspberrypi/userland #216: VC_IMAGE_1BPP needs a palette?

ali1234, 2015-02-06:
I attempted to use 1BPP image format. What I found is that the image gets rendered as black and very dark blue instead of black and white like I would expect.

I also tried to set a palette on the image with vc_dispmanx_resource_set_palette as per #95 but this had no effect on the output.

Ruffio, 2015-06-27:
@ali1234 is this still an issue?

ali1234, 2015-07-14:
Yes.

6by9, 2015-07-14:
1BPP appears to be a palettised format from the lowest level hardware driver, but vc_dispmanx_resource_set_palette says it is only for 4BPP and 8BPP formats.
The GPU code for vc_dispmanx_resource_set_palette checks for the palette being a block of either 16 (4BPP) or 256 (8BPP) 16 or 32 bit words, and the image manipulation library likewise seems to really only have handling for 4BPP and 8BPP.
It looks like 1BPP support is ropey to say the least, but I don't know enough about how it was intended to be supported to look further.

ali1234, 2015-07-14:
My assumption is that the dark blue colour comes from treating the "1"s in the 1BPP image as a colour and casting it to 16 or 32 bit, resulting in only the lowest bit of the blue channel being turned on.

I once got a 1BPP surface to display with a palette, and also crashed the VC in the process. I have two versions of my code for 1BPP and 8BPP but I forgot to change the pitch so it overran the buffers and crashed, leaving the 1BPP surface behind - but I didn't immediately notice this. When I next ran the 8BPP version it's surface was not displayed but the palette seemed to be applied to the crashed 1BPP surface somehow. Then I had to reboot to get things working again.

JamesH65, 2017-12-05:
Its seems unlikely that we will be making any changes to this area to support this (cost benefit is negative). So closing as a wont fix.

closed 2017-12-05 by JamesH65

nistvan86, 2020-08-18:
Is there any way (even a hack with the gencmd API) to set the pixel color in 1BPP mode? In my PiCM renderer (https://github.com/nistvan86/picm) I was planning to use this mode to directly copy binary data to the GPU memory, which is a lot faster than converting each bit to 4bpp/8bpp bytes.

ali1234, 2020-08-18:
I never found a way to make it work.

popcornmix, 2020-08-18:
You may be better off trying to use the arm-side kms driver for this rather than firmware driver.
If it's not supported then the code is modifiable.
The hardware appears to support 1, 2, 4, and 8 bpp palettised formats.

6by9, 2020-08-18:
I don't think the vc4 driver supports palettes. HVS_PIXEL_FORMAT_PALETTE is defined in vc4_regs.h, but there's no support code that uses it.