raspberrypi/userland #220: vcos_timer_delete and other functions incorrectly declared inline

rcombs, 2015-02-16:
This produces giant piles of compiler warnings:
```
/opt/vc/include/interface/vcos/vcos_timer.h:112:6: warning: inline function ‘vcos_timer_delete’ declared but never defined [enabled by default]
/opt/vc/include/interface/vcos/vcos_timer.h:109:6: warning: inline function ‘vcos_timer_reset’ declared but never defined [enabled by default]
/opt/vc/include/interface/vcos/vcos_timer.h:102:6: warning: inline function ‘vcos_timer_cancel’ declared but never defined [enabled by default]
/opt/vc/include/interface/vcos/vcos_timer.h:95:6: warning: inline function ‘vcos_timer_set’ declared but never defined [enabled by default]
/opt/vc/include/interface/vcos/vcos_timer.h:80:15: warning: inline function ‘vcos_timer_create’ declared but never defined [enabled by default]
/opt/vc/include/interface/vcos/vcos_mem.h:93:15: warning: inline function ‘vcos_get_free_mem’ declared but never defined [enabled by default]
/opt/vc/include/interface/vcos/vcos_thread.h:269:6: warning: inline function ‘vcos_thread_resume’ declared but never defined [enabled by default]
/opt/vc/include/interface/vcos/vcos_thread.h:262:5: warning: inline function ‘vcos_thread_running’ declared but never defined [enabled by default]
/opt/vc/include/interface/vcos/vcos_thread.h:251:15: warning: inline function ‘vcos_change_preemption’ declared but never defined [enabled by default]
/opt/vc/include/interface/vcos/vcos_thread.h:234:6: warning: inline function ‘vcos_thread_relinquish’ declared but never defined [enabled by default]
/opt/vc/include/interface/vcos/vcos_thread.h:186:15: warning: inline function ‘vcos_thread_get_affinity’ declared but never defined [enabled by default]
```

popcornmix, 2015-02-16:
Are you building on Pi or cross compiling?
Debug or release build?

rcombs, 2015-02-16:
Building on the Pi; debug build. The relevant functions are defined in here:
https://github.com/raspberrypi/userland/blob/4333d6d023a1f0dcc763d0706a1fe6523f9b6005/interface/vcos/pthreads/vcos_pthreads.c#L874
but declared as inline in here:
https://github.com/raspberrypi/userland/blob/4062ce5b2492b4eae4bb9816a7a43c00f7a216de/interface/vcos/vcos_timer.h#L112

asavah, 2015-05-03:
This gets even worse with gcc 5.1.0 (cross, crosstool-ng based,  Release build)

In file included from ~/src/vc4/build/inc/interface/vcos/vcos.h:127:0,
                 from ~/src/vc4/interface/vcos/pthreads/vcos_pthreads.c:29:
~/src/vc4/build/inc/interface/vcos/vcos_thread.h:269:6: error: inline function ‘vcos_thread_resume’ declared but never defined [-Werror]
 void vcos_thread_resume(VCOS_THREAD_T *thread);
      ^
~/src/vc4/build/inc/interface/vcos/vcos_thread.h:262:5: error: inline function ‘vcos_thread_running’ declared but never defined [-Werror]
 int vcos_thread_running(VCOS_THREAD_T *thread);
     ^
~/src/vc4/build/inc/interface/vcos/vcos_thread.h:251:15: error: inline function ‘vcos_change_preemption’ declared but never defined [-Werror]
 VCOS_UNSIGNED vcos_change_preemption(VCOS_UNSIGNED pe);
               ^
~/src/vc4/build/inc/interface/vcos/vcos_thread.h:234:6: error: inline function ‘vcos_thread_relinquish’ declared but never defined [-Werror]
 void vcos_thread_relinquish(void);
      ^
~/src/vc4/build/inc/interface/vcos/vcos_thread.h:186:15: error: inline function ‘vcos_thread_get_affinity’ declared but never defined [-Werror]
 VCOS_UNSIGNED vcos_thread_get_affinity(VCOS_THREAD_T *thread);
               ^

(Substituted the full /home/blah/path by "~/")

Ruffio, 2015-06-27:
@rcombs @asavah is this still an issue?

kr15h, 2015-06-29:
Its still an issue for me. Just cloned the latest version and got this while compiling on Arch Linux ARM:
```bash
In file included from /home/kerberos-io/build/userland/build/inc/interface/vcos/vcos.h:127:0,
                 from /home/kerberos-io/build/userland/interface/vcos/pthreads/vcos_pthreads.c:29:
/home/kerberos-io/build/userland/build/inc/interface/vcos/vcos_thread.h:269:6: error: inline function 'vcos_thread_resume' declared but never defined [-Werror]
 void vcos_thread_resume(VCOS_THREAD_T *thread);
      ^
/home/kerberos-io/build/userland/build/inc/interface/vcos/vcos_thread.h:262:5: error: inline function 'vcos_thread_running' declared but never defined [-Werror]
 int vcos_thread_running(VCOS_THREAD_T *thread);
     ^
/home/kerberos-io/build/userland/build/inc/interface/vcos/vcos_thread.h:251:15: error: inline function 'vcos_change_preemption' declared but never defined [-Werror]
 VCOS_UNSIGNED vcos_change_preemption(VCOS_UNSIGNED pe);
               ^
/home/kerberos-io/build/userland/build/inc/interface/vcos/vcos_thread.h:234:6: error: inline function 'vcos_thread_relinquish' declared but never defined [-Werror]
 void vcos_thread_relinquish(void);
      ^
/home/kerberos-io/build/userland/build/inc/interface/vcos/vcos_thread.h:186:15: error: inline function 'vcos_thread_get_affinity' declared but never defined [-Werror]
 VCOS_UNSIGNED vcos_thread_get_affinity(VCOS_THREAD_T *thread);
               ^
```

fabled, 2015-08-27:
The macro definitions are incorrect in vcos module. As workaround I added to my build the patch:
http://git.alpinelinux.org/cgit/aports/tree/main/raspberrypi/fix-inlines.patch

I also need the following patch to build against musl:
http://git.alpinelinux.org/cgit/aports/tree/main/raspberrypi/fix-build.patch

zelmon64, 2015-10-06:
I had this issue building on a raspberry pi 2 (jessie) but solved it by doing "sudo make install" rather than just "make install". I built it with cmake 3.0.2 by changing the minimum required version to 3.0 and adding the line "SET(CMAKE_C_FLAGS "-std=gnu99")" to the CMakeLists.txt file.

idreamerhx, 2015-12-25:
fix-inlines.patch solved compiling on archlinux raspberr pi. thanks.

6by9, 2016-01-01:
Change merged to fix this.
The definitions are correct - VCOS is an abstraction layer and on most platforms that abstraction is done with inlines. Pthreads needs a bit more work, so had actual function implementations. Inlines now defined to redirect to the actual implementations.
Missing functions defined as stubs as they are not applicable to pthreads.

Please retest, and close if happy.

6by9, 2016-01-16:
@rcombs @kr15h Please review fixed code. Whilst there are still warnings from the code, the inline issue should be resolved.

asavah, 2016-01-21:
@6by9
Cross-builds with gcc-5.3.0 and gcc-linaro-snapshot-5.3-2016.01 (crosstool-ng based toolchain) succeeded for me.

popcornmix, 2016-06-03:
I believe this has been resolved. Feel free to reopen if it hasn't.

closed 2016-06-03 by popcornmix