raspberrypi/userland #221: Redundant redefinition warnings in vcos_blockpool

rcombs, 2015-02-16:
```
/opt/vc/include/interface/vcos/vcos_blockpool.h:69:15: warning: redundant redeclaration of ‘vcos_blockpool_init’ [-Wredundant-decls]
/opt/vc/include/interface/vcos/vcos_blockpool.h:142:10: warning: redundant redeclaration of ‘vcos_blockpool_elem_to_handle’ [-Wredundant-decls]
/opt/vc/include/interface/vcos/generic/vcos_generic_blockpool.h:209:15: note: previous definition of ‘vcos_blockpool_init’ was here
/opt/vc/include/interface/vcos/generic/vcos_generic_blockpool.h:271:10: note: previous definition of ‘vcos_blockpool_elem_to_handle’ was here
/opt/vc/include/interface/vcos/vcos_blockpool.h:147:7: warning: redundant redeclaration of ‘vcos_blockpool_elem_from_handle’ [-Wredundant-decls]
/opt/vc/include/interface/vcos/vcos_blockpool.h:91:15: warning: redundant redeclaration of ‘vcos_blockpool_create_on_heap’ [-Wredundant-decls]
/opt/vc/include/interface/vcos/generic/vcos_generic_blockpool.h:277:7: note: previous definition of ‘vcos_blockpool_elem_from_handle’ was here
/opt/vc/include/interface/vcos/generic/vcos_generic_blockpool.h:219:15: note: previous definition of ‘vcos_blockpool_create_on_heap’ was here
/opt/vc/include/interface/vcos/vcos_blockpool.h:152:10: warning: redundant redeclaration of ‘vcos_blockpool_is_valid_elem’ [-Wredundant-decls]
/opt/vc/include/interface/vcos/vcos_blockpool.h:103:7: warning: redundant redeclaration of ‘vcos_blockpool_alloc’ [-Wredundant-decls]
/opt/vc/include/interface/vcos/generic/vcos_generic_blockpool.h:283:10: note: previous definition of ‘vcos_blockpool_is_valid_elem’ was here
/opt/vc/include/interface/vcos/vcos_blockpool.h:165:18: warning: redundant redeclaration of ‘vcos_blockpool_extend’ [-Wredundant-decls]
/opt/vc/include/interface/vcos/generic/vcos_generic_blockpool.h:228:28: note: previous definition of ‘vcos_blockpool_extend’ was here
/opt/vc/include/interface/vcos/generic/vcos_generic_blockpool.h:235:7: note: previous definition of ‘vcos_blockpool_alloc’ was here
/opt/vc/include/interface/vcos/vcos_blockpool.h:112:7: warning: redundant redeclaration of ‘vcos_blockpool_calloc’ [-Wredundant-decls]
/opt/vc/include/interface/vcos/generic/vcos_generic_blockpool.h:241:7: note: previous definition of ‘vcos_blockpool_calloc’ was here
/opt/vc/include/interface/vcos/vcos_blockpool.h:119:6: warning: redundant redeclaration of ‘vcos_blockpool_free’ [-Wredundant-decls]
/opt/vc/include/interface/vcos/generic/vcos_generic_blockpool.h:247:6: note: previous definition of ‘vcos_blockpool_free’ was here
/opt/vc/include/interface/vcos/vcos_blockpool.h:125:18: warning: redundant redeclaration of ‘vcos_blockpool_available_count’ [-Wredundant-decls]
/opt/vc/include/interface/vcos/generic/vcos_generic_blockpool.h:253:15: note: previous definition of ‘vcos_blockpool_available_count’ was here
/opt/vc/include/interface/vcos/vcos_blockpool.h:131:18: warning: redundant redeclaration of ‘vcos_blockpool_used_count’ [-Wredundant-decls]
/opt/vc/include/interface/vcos/generic/vcos_generic_blockpool.h:259:15: note: previous definition of ‘vcos_blockpool_used_count’ was here
/opt/vc/include/interface/vcos/vcos_blockpool.h:138:6: warning: redundant redeclaration of ‘vcos_blockpool_delete’ [-Wredundant-decls]
/opt/vc/include/interface/vcos/generic/vcos_generic_blockpool.h:265:6: note: previous definition of ‘vcos_blockpool_delete’ was here
/opt/vc/include/interface/vcos/vcos_blockpool.h:142:10: warning: redundant redeclaration of ‘vcos_blockpool_elem_to_handle’ [-Wredundant-decls]
/opt/vc/include/interface/vcos/generic/vcos_generic_blockpool.h:271:10: note: previous definition of ‘vcos_blockpool_elem_to_handle’ was here
/opt/vc/include/interface/vcos/vcos_blockpool.h:147:7: warning: redundant redeclaration of ‘vcos_blockpool_elem_from_handle’ [-Wredundant-decls]
/opt/vc/include/interface/vcos/generic/vcos_generic_blockpool.h:277:7: note: previous definition of ‘vcos_blockpool_elem_from_handle’ was here
/opt/vc/include/interface/vcos/vcos_blockpool.h:152:10: warning: redundant redeclaration of ‘vcos_blockpool_is_valid_elem’ [-Wredundant-decls]
/opt/vc/include/interface/vcos/generic/vcos_generic_blockpool.h:283:10: note: previous definition of ‘vcos_blockpool_is_valid_elem’ was here
/opt/vc/include/interface/vcos/vcos_blockpool.h:165:18: warning: redundant redeclaration of ‘vcos_blockpool_extend’ [-Wredundant-decls]
/opt/vc/include/interface/vcos/generic/vcos_generic_blockpool.h:228:28: note: previous definition of ‘vcos_blockpool_extend’ was here
```

I believe the solution should be as simple as adding an `#include "interface/vcos/vcos_blockpool.h"` at https://github.com/raspberrypi/userland/blob/4062ce5b2492b4eae4bb9816a7a43c00f7a216de/interface/vcos/generic/vcos_generic_blockpool.h#L46 to make sure the declarations are parsed before the implementations.

Ruffio, 2015-06-27:
@rcombs is this still an issue?

popcornmix, 2016-06-03:
I believe this has been resolved. Feel free to reopen if it hasn't.

closed 2016-06-03 by popcornmix