raspberrypi/userland #224: Small typo in RaspiTexUtil.c

eliotn, 2015-03-02:
in raspitexutil_gl_term(RASPITEX_STATE *raspitex_state):

   glDeleteTextures(1, &raspitex_state->v_texture);
   eglDestroyImageKHR(raspitex_state->display, raspitex_state->v_egl_image);
   raspitex_state->u_egl_image = EGL_NO_IMAGE_KHR;

In the third line u_egl_image was probably intended to be v_egl_image, otherwise its a no-op operation as raspitex_state->u_egl_image was set 4 lines up.

popcornmix, 2015-03-03:
Thanks. You are correct.
The typo is harmless in this case, but no typo is better.
I've fixed it in our internal tree - will appear in next update.

popcornmix, 2015-03-08:
Fixed here: https://github.com/raspberrypi/userland/commit/041604e47e73fb2a9a487ce6f446e9b2d2042546

closed 2015-03-08 by popcornmix