raspberrypi/userland #227: vp8 encoding for raspivid (for webm, html5 video streaming support)

maxogden, 2015-03-13:
I think it would really useful to have VP8 encoding support in `raspivid`. I am not very good at C/C++ but I looked into MMAL and it seems to support VP8. 

The benefit of support VP8 (rather than H264) is that you can stream video directly from raspivid into a web browser `<video>` tag with webm, whereas mp4 can't stream.

I would normally just transcode the H264 into VP8, but I can only get 2-3 FPS at 1080p on the Raspberry Pi A+, whereas I get 25FPS from Raspivid normally at 1080P.

Does anyone know how hard it would be to add VP8 support?

popcornmix, 2015-03-14:
There is no hardware support for VP8, and no software VP8 encoder, so this is not going to happen.

closed 2015-03-14 by popcornmix

zoff99, 2018-05-06:
@popcornmix what do you mean with "no software VP8 encoder" ?

6by9, 2018-05-06:
raspivid only uses the GPU firmware. There is no software VP8 software encoder available in the firmware.
It would potentially be possible to use FFmpeg or similar on the ARM cores of a Pi 2 or 3 to encode to VP8, but there is no guarantee on the performance that can be achieved, and it is not functionality that will get added to raspivid.