raspberrypi/userland #228: Potential thread safety issues with MMAL

wm4, 2015-03-16:
I believe I found an issue in mmal_queue. Here's its implementation:

```C
MMAL_BUFFER_HEADER_T *mmal_queue_get(MMAL_QUEUE_T *queue)
{
   MMAL_BUFFER_HEADER_T *buffer;

	if(!queue) return 0;

   vcos_mutex_lock(&queue->lock);
   buffer = queue->first;
   if(!buffer)
   {
      vcos_mutex_unlock(&queue->lock);
      return 0;
   }

   /* coverity[lock] This semaphore isn't being used as a mutex */
   vcos_semaphore_wait(&queue->semaphore); /* Will always succeed */

   queue->first = buffer->next;
   if(!queue->first) queue->last = &queue->first;

   queue->length--;
   vcos_mutex_unlock(&queue->lock);

   return buffer;
}

MMAL_BUFFER_HEADER_T *mmal_queue_wait(MMAL_QUEUE_T *queue)
{
	if(!queue) return 0;

	vcos_semaphore_wait(&queue->semaphore);
   vcos_semaphore_post(&queue->semaphore);
   return mmal_queue_get(queue);
}
```

What happens if 2 threads calls ``mmal_queue_wait()`` at about the same time? Thread 1 could be preempted right after calling ``vcos_semaphore_post()``, during which thread 2 successfully runs the same code and dequeues a buffer. Now assume the queue size was 1, and thread 2 just reduced it to 0. Thread 1 already verified that the queue size was 1 (which is now outdated), and would return NULL instead of waiting. (This could be easily avoided by using a condition variable instead of a semaphore. Or the cheap and dumb solution: loop if the result is NULL.)

I found another potential issue with buffers. This is how the ``mmal_buffer_header_release()`` is implemented:
```C
void mmal_buffer_header_release(MMAL_BUFFER_HEADER_T *header)
{
#ifdef ENABLE_MMAL_EXTRA_LOGGING
   LOG_TRACE("%p (%i)", header, (int)header->priv->refcount-1);
#endif

   if(--header->priv->refcount != 0)
      return;

  [...]
```

As you can see, there is no lock around changing and checking ``refcount``. This makes the function not thread-safe. Yet, ``example_basic_1.c`` uses it in a context where threading is used:

```C
static void input_callback(MMAL_PORT_T *port, MMAL_BUFFER_HEADER_T *buffer)
{
   struct CONTEXT_T *ctx = (struct CONTEXT_T *)port->userdata;

   /* The decoder is done with the data, just recycle the buffer header into its pool */
   mmal_buffer_header_release(buffer);

   /* Kick the processing thread */
   vcos_semaphore_post(&ctx->semaphore);
}
```

Now, maybe this is actually fine, because you could argue that the buffer is owned only by 1 thread, and referencing a buffer in another thread is not allowed. So the example happens to use this in a safe way. But then, why have refcounting?

Is this a bug? If not, could thread-safety be explicitly documented?

(Just by the way, I'm trying to implement video decoding with the mmal API. Although the API is pretty simple, I'm still struggling to make it stable. For example, it tends to randomly freeze the whole rpi depending on exact timing, and inserting random ``usleep(1000)`` in my code "fixes" it to some degree. This happens only with some files, while other files work perfectly fine. omxplayer never has problems. I'm still trying to find out why.)

popcornmix, 2015-03-16:
ping @6by9

6by9, 2015-03-16:
I can't think of a sane MMAL use case where you want two threads waiting to receive a buffer from the same queue. 
Completions aren't available on the GPU, and this code is built on both ARM and GPU sides. Looping until not-NULL would be a solution as you say.

I'd need to have a think on mmal_buffer_header_release, but I think you may be correct.
The reason for the refcounting is that it was nominally a simple way to implement a splitter - receive a buffer, inc the refcount (via acquire), and send it on to 2 sinks. Only when both sinks release the header will it be returned to the pool, and the release callback on the pool can then send the buffer back to the source.
I've never used it in that way, but that was the original intent.

I take it you have found that you can turn logging on within MMAL. 
https://github.com/raspberrypi/userland/blob/master/interface/mmal/core/mmal_logging.c#L31 change VCOS_LOG_ERROR to VCOS_LOG_TRACE and it'll spit out a lot more tracing.

wm4, 2015-03-16:
>I can't think of a sane MMAL use case where you want two threads waiting to receive a buffer from the same queue. 

It should be documented as unsafe, then.

>I take it you have found that you can turn logging on within MMAL.
>https://github.com/raspberrypi/userland/blob/master/interface/mmal/core/mmal_logging.c#L31 change VCOS_LOG_ERROR to VCOS_LOG_TRACE and it'll spit out a lot more tracing.

No, I haven't found this yet. Is there a way to do this without rebuilding firmware? (It looks like it's impossible, unless the mmal shared lib happens to export mmal_log_category.)

wm4, 2015-03-16:
>(It looks like it's impossible, unless the mmal shared lib happens to export mmal_log_category.)

Oh, I'm blind... it does.

pelwell, 2015-03-16:
There is an easier way - setting the VC_LOGLEVEL environment variable to "mmal:trace" should do the trick.

[In the general case it takes a comma separated list of "name:level" pairs.

6by9, 2015-03-16:
Ah, magic even I didn't know of. Thanks Phil.

Ruffio, 2015-06-27:
@wm4 is this still an issue?

wm4, 2015-06-27:
It's a theoretical issue. I haven't checked whether there were changes to this part of userland, but if there were no changes, I'd claim it still exists.

Anyway, I finished my MMAL decoding implementation. It's in FFmpeg. It seems to work sufficiently so far. Saner MMAL_BUFFER_HEADER_T behavior (such as being able to reference a buffer even if the pool is destroyed) might have been useful, but I worked it around successfully.

Ruffio, 2016-12-30:
@6by9 @wm4 is this a 'real' issue and should be fixed or should it just be closed?

popcornmix, 2016-12-30:
I think https://github.com/raspberrypi/userland/commit/ab0b1c1839d931d2a1a10da6ef5ca255de57e3d3 may fix the thread safety issue with two receive threads.

JamesH65, 2017-12-05:
Closing due to lack of activity. Reopen if you feel this issue is still relevant. Appears it may be fixed anyway, if not please reopen.

closed 2017-12-05 by JamesH65