raspberrypi/userland #229: Simple function to wait for vsync without submitting an update

vanfanel, 2015-03-18:
Hi,

I have been looking in the VideoCore includes but could not find a function that does simply this: block until next vsync arrives. Very easy to implement, and would only need a display handle (DISPMANX_DISPLAY_HANDLE_T) as parameter. Something as simple as this:

vc_dispmanx_wait_vsync(DISPMANX_DISPLAY_HANDLE_T display);

Until now, I've been using the old fbdev interface and corresponding WAITFORVYNC ioctl for this, but it's a hacky solution.
Could it be done, please?

popcornmix, 2015-03-18:
You can already register for a callback every vsync with vc_dispmanx_vsync_callback. See:
https://github.com/popcornmix/xbmc/commit/f53bf236b8892baf25b3879256037b6070025fb9

vanfanel, 2015-03-21:
Oh, true!
Closing this, popcornmix. Thanks a lot, as always :+1:

closed 2015-03-21 by vanfanel