raspberrypi/userland #230: Decoding in lockstep with MMAL

wm4, 2015-03-18:
I want to add a MMAL decoder wrapper to FFmpeg. FFmpeg's API requires that  output frame for each input packet is produced. The API user must pass a packet containing exactly one video frame to the decode function, and gets back a decoded frame. To support b-frames and multithreaded decoding, there can be a delay with which decoded frames are returned: the first N decode calls will not return decoded frames, but in exchange you must call the decode function N times without a packet to "flush" the decoder.

The MMAL API is asynchronous and quite different. You keep pushing in packets (until the input buffer is exhausted, I guess), and the decoder will return decoded frames at completely independent from this.

When connecting these two APIs, there is one main problem: how do you know after how many packets you can expect an output frame from the decoder? This is needed to determine the minimum value for N. Basically: once the user has fed N packets to the decoder, the decoder must return an output frame for each packet. (And wait for the hw decoder if no output frame is ready yet.)

You could choose N based on when the first video frame happens to be decoded. This makes N basically random (based on the exact timings). The worst about it is that it can lead to excessive buffering, and a "logic" deadlock as a consequence: the decoder can suddenly return a large number of frames in a burst. After this, it won't consume more input until it has fed new video frames through the output port. But the FFmpeg wrapper can't return these frames quickly enough to the user (it can return 1 frame for 1 input packet). The input queue quickly fills up, the input pool gets exhausted, no output frames can be recycled, and everything locks up. Basically, you have a full output queue and empty input/output pools. I plan to work this around by queuing input packets in RAM using a queue outside of the MMAL API, but the point is that there's too much buffering.

So, to make it short: is there a way to determine when the MMAL component was fed enough input that we can expect output? I want to wait on the decoder, but not forever.

I've also managed to freeze the GPU completely quite often. The CPU still worked, but every single process trying to access VCOS went into uninterruptible sleep. I suspect this was due to out of memory conditions on the GPU. But I can't know for sure.  If this is true, then this excessive buffering is quite dangerous and I absolutely have to avoid it.

(Sorry if such API questions are inappropriate on this issue tracker.)

6by9, 2015-03-19:
Ping @deborah-c for confirmation on what I'm about to say.

Firstly, how are you handling header bytes? Those typically get sent as flagged buffers but will produce no output. Does N always have to be 1 to cope with that, or do you get given the header bytes via some other route? What about dynamic resolution changes which require a new set of header bytes?

Secondly it's going to depend on the stream.
If you have a stream with no b-frames, then (ignoring header bytes) it will be one in, one out as you have all the reference frames already.
If your decoded sequence is IBPBPBP, then MMAL will give you a frame after the I, have to present the P but get no output, present the B and get an output, and then you'll get the P frame delivered. So N can be 1, and you'll have to delay delivering the I frame output by one frame.
If your decoded sequence is IBBPBBPBBP, then N needs to be 2. etc.

I'm not even certain that the header bytes will give this information. They should indicate how many reference frames are used, but not the direction of them.
If that is the ffmpeg API, then unless you can deduce that information from the stream, I don't see a solution other than setting N for a worst case.

wm4, 2015-03-19:
>Firstly, how are you handling header bytes? Those typically get sent as flagged buffers but will produce no output. Does N always have to be 1 to cope with that, or do you get given the header bytes via some other route?

With the FFmpeg API, you first have to send some input packets before you get the first frame. The number of packets is N. It's fully up to the decoder what the value of N is. Usually it's the number of frames that must be buffered due to b-frame reordering, plus an additional delay to make multithreaded decoding work. The upper bound on N in the h264 case is probably ``16 + number_of_threads``.

>What about dynamic resolution changes which require a new set of header bytes?

This is fully transparent. The value of N might change; the API can't lower the value of N, but it can increase it by just not returning frames for a few packets. (Unusual, but possible.)

>Secondly it's going to depend on the stream.

Yep. Basically, it should be internal to the decoder. In FFmpeg's h264 decoder for example, even if you have a stream with no b-frames, N would be larger than 1 if frame-threading is enabled. You could call this "forced buffering", and it's unclear to me whether MMAL has it.

Basically, I either need:

a) a good upper bound for N which will work with all mmal decoders
b) a way to determine whether the decoder is still working on an output image, or whether it's actually waiting for new input

With b) I could just do the following on initialization:
1. Send a packet
2. Check the decoder state: 
    1) if it's still decoding, wait
    2) if not, proceed to 3
3. Check if there's an output frame and possibly return it.

This will prevent the deadlock case, because the difference between sent input and returned output frames can't get too big. (If it's too big, we can't return the output frames to the user, which will lead to the output pool being exhausted and the output queue being full, which in turn blocks the decoder, which in turn exhausts the input pool, which in turn blocks the ffmpeg wrapper.)

>I don't see a solution other than setting N for a worst case.

The question is, what is the worst case?

The following has to work:

1. Send N input packets (each containing a compressed video frame).
2. After this, for each input packet an output frame is read.
3. If sending an input packet doesn't yield an output frame immediately, it will block until an output frame arrives (this must exclude the possibility of deadlocks - I'm not going to add timeouts either).

(I repeated this a few times, but it's probably also not easy to understand why exactly this is a problem or why it happens.)

>I'm not even certain that the header bytes will give this information. They should indicate how many reference frames are used, but not the direction of them.

Yes, it would probably be possible to parse the packets myself to determine the stream properties. Seems like it'd be overly complex, though. It doesn't account for "forced buffering" in the MMAL decoder either.

JamesH65, 2015-03-19:
Can you use the buffer calbacks on input and output ports to count and
record what is going on? The flags returned with each buffer indicate
whether the buffer is at the end of the frame etc.

wm4, 2015-03-19:
>Can you use the buffer calbacks on input and output ports to count and record what is going on? 

This is not enough. I can only detect a deadlock this way when it's too late.

To reiterate the problem again: when I push input frames to the decoder, I don't know when I can stop and wait for output frames. I could do it until an output frame happens to arrive - but this happens at an essentially random print in time because of the asynchronicity. This on the other hand can lead to too much buffering, which will lead to a deadlock situation if I lockstep packet input/frame output is done.

JamesH65, 2015-03-19:
Why can't you just keep pushing until a frame is returned? And how do you
know you are pushing an entire frame in? Is the H264 stream already parsed
to that extent?

wm4, 2015-03-19:
>Why can't you just keep pushing until a frame is returned?

Because then I run into the deadlock: the decoder will consume buffers from the output pool until it's exhausted. Then it will stop accepting new input. The input pool will exhaust too. Why can't I just use the output buffers? Because I can return only 1 output buffer per 1 input packet. The problem is made worse by the fact that 1 input packet may require multiple mmal input buffers (because they're fixed sized for some reason).

>Is the H264 stream already parsed to that extent?

Yes. It's a requirement of the FFmpeg API, and FFmpeg demuxers already output packets in this format. Many file formats also have this requirement (like mkv, avi, mp4).

6by9, 2015-03-19:
MMAL buffers aren't fixed size. You can alter the buffer_size parameter in the port format up until the point where you enable the port. Just because (for example) https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiVid.c#L1486 takes buffer_size_recommended (defaults to 64kB IIRC for encoded video), there's nothing stopping you making it 512kB or more to guarantee you should be able to fit a single frame in (had similar fun with a different project where the encode path wouldn't allow fragmented buffers. IIRC we ended up with 768kB buffers for 1080P, and 512kB for up to 720P. Predicting max frame size when doing realtime encoding really isn't easy!)

Are you trying to do realtime decode, or as fast as possible for transcode? Do you get to see timestamps for the frames?
I would be surprised if a stream ever needed N to exceed 5 (assuming we're ignoring header bytes). Maintain a limited size queue, and if you've hit the limit then wait. It'd also maximise the codec utilisation, as it won't be waiting around for new data to consume.

wm4, 2015-03-19:
>MMAL buffers aren't fixed size. You can alter the buffer_size parameter in the port format up until the point where you enable the port.

Sure, but during decoding they are essentially fixed. I don't think I can just set a buffer to the whole FFmpeg packet; I have to slice it up and feed it to the decoder incrementally.

However, I could just use a FFmpeg buffer pool instead of MMAL's API to allocate buffers. Then I could just allocate more buffer headers on the fly if needed. This is probably the most elegant way to solve this.

wm4, 2015-03-19:
Oh, allow me to ask another question, probably slightly offtopic: does a MMAL_ENCODING_OPAQUE buffer stay valid even after the decoder component has been destroyed?

6by9, 2015-03-19:
You can allocate the port format buffer_size to any size above buffer_size_min, copy your data in to the start of the buffer, and then set the ```length``` field in the ```MMAL_BUFFER_HEADER_T``` correctly to describe the amount of data you copied in. You already said that FFMPEG gave you the correctly demuxed data packets.
I'd also recommend you set the buffer flags to include ```MMAL_BUFFER_HEADER_FLAG_FRAME_END``` as that means the codec isn't having to hunt for start codes.

Yes, a MMAL_ENCODING_OPAQUE buffer should remain valid after the component has been destroyed assuming it was never returned to the component. Be warned that it keeps the full image pool allocated until the last reference is released, so you may be using more memory than ideal.

wm4, 2015-03-19:
>You can allocate the port format buffer_size to any size above buffer_size_min, copy your data in to the start of the buffer, and then set the length field in the MMAL_BUFFER_HEADER_T correctly to describe the amount of data you copied in. 

Yeah, but you don't know the upper bound of the packet. I suppose it's not valid to set buffer_size to SIZE_MAX? (I wouldn't actually allocate the buffer, but make it point to the FFmpeg buffer allocation.)

>I'd also recommend you set the buffer flags to include MMAL_BUFFER_HEADER_FLAG_FRAME_END as that means the codec isn't having to hunt for start codes.

Thanks for the hint. I will try it.

>Yes, a MMAL_ENCODING_OPAQUE buffer should remain valid after the component has been destroyed assuming it was never returned to the component. Be warned that it keeps the full image pool allocated until the last reference is released, so you may be using more memory than ideal. 

Does image pool here refer to the MMAL_POOL_T (which I probably won't use in favor of a FFmpeg pool), or something hidden allocated in MMAL or on videocore?

6by9, 2015-03-19:
Using FFMpeg's buffer allocation? So how are you passing that data via MMAL? You've got to have a set of MMAL_BUFFER_HEADER_Ts which will then also reference some GPU allocated memory. If you're not  using a MMAL_POOL_T, then you're going to have to post your code somewhere as it sounds like you're doing something outside the normal use.
Upper bound of the packet: you have to go for a worst case, hence my mention of 512kB or larger. 

Underlying image pool on Videocore. Video decode (and numerous other source components) will allocate GPU memory to take the actual images. OPAQUE buffers are then passing reference counted handles to those buffers. The buffer itself holds a reference which is only released on the buffer being returned to the source component, or on the buffer being deallocated. 

I suspect that opaque buffers are NOT the way you want to go to start with. As julian who has been integrating MMAL into VLC will attest, they complicate matters over exactly what they reference, and are only of benefit if the ARM never wants to see the pixel content. Seeing as FFMPEG is often used for transcoding, I'd say that is not worth it.

wm4, 2015-03-19:
> Using FFMpeg's buffer allocation? So how are you passing that data via MMAL? You've got to have a set of MMAL_BUFFER_HEADER_Ts which will then also reference some GPU allocated memory. If you're not [...]

I mean I will be using some FFmpeg data structures to manage the MMAL_BUFFER_HEADER_Ts, instead of a MMAL_POOL_T. This would give me more flexibility. For example, I want references to decoded images to be valid even if the decoder is destroyed. This is transparent with FFmpeg's refcounted buffers, and destroying the FFmpeg pool will only delete unused buffers.

>Upper bound of the packet: you have to go for a worst case, hence my mention of 512kB or larger. 

But the upper bound can't be known without scanning the whole file in advance.

> Underlying image pool on Videocore. Video decode (and numerous other source components) will allocate GPU memory to take the actual images. OPAQUE buffers are then passing reference counted handles [...]

So keeping a reference to a single buffer keeps the complete pool alive? This is bad for me.

> I suspect that opaque buffers are NOT the way you want to go to start with. As julian who has been integrating MMAL into VLC will attest, they complicate matters over exactly what they reference, [...]

I'm using it for playback. It already works, but I still need to fix some unsolved problems over buffer references. FFmpeg uses refcounted frames, and it's allowed to keep references to them after the decoder is destroyed.

wm4, 2015-03-19:
> Using FFMpeg's buffer allocation? So how are you passing that data via MMAL? You've got to have a set of MMAL_BUFFER_HEADER_Ts which will then also reference some GPU allocated memory. If you're not [...]

I realized that I can't allocate single MMAL_BUFFER_HEADER_Ts (only pools of it) using the API, so I abandoned this idea. Instead I'm keeping an extra list of "unconsumed" input buffers, and I'm refcounting the pool as a whole.

In case someone wants to take a look, here's my ffmpeg patch: https://github.com/FFmpeg/FFmpeg/compare/master...wm4:rpi (it got rather wild, although some of it is unneeded debugging stuff)
The player part which uses this API: https://github.com/mpv-player/mpv/compare/master...wm4:rpi

6by9, 2015-03-20:
You could allocate MMAL_BUFFER_HEADER_Ts, but calls such as mmal_buffer_header_release will object as they rely on the buffer being allocated from a pool. MMAL_POOL_Ts are the intended mechanism for handling buffers.

I've had a quick look at your code. It looks like you're doing a few sneaky tricks (which I think I did in the past too!) but haven't seen anything particularly hideous other than not knowing the FFMPEG API.
I don't believe EGL will be able to consume MMAL_ENCODING_OPAQUE packets, but as you've got a MMAL renderer too I'm not sure if that is being used.

mmaldec.c line 399 I recommend you go for a buffer size of 512kB rather than buffer_size_recommended. It'll avoid most of your issues having to split a frame into multiple MMAL buffers. Make buffer_num something suitable and you should be able to avoid your intermediate buffers totally, and just memcpy into the MMAL buffers in mmal_add_packet. Seeing as you appear to be already copying it twice within add_packet (I don't understand the if (avpkt && avpkt->data && avpkt->size) clause - is that the normal path for data?) it might make things simpler. I can't say for sure as I don't know the API.

I'll try to have a further look later (got to do something whilst waiting for stuff to compile)

wm4, 2015-03-20:
Thanks for having a look, I really appreciate it.

>I don't believe EGL will be able to consume MMAL_ENCODING_OPAQUE packets

This is what I've tried first, but I couldn't get it to work. Raspivid seems to include code for this, though. But not using the decoder component apparently. Well, the device is probably too weak for this anyway.

>mmaldec.c line 399 I recommend you go for a buffer size of 512kB rather than buffer_size_recommended. It'll avoid most of your issues having to split a frame into multiple MMAL buffers.

Right, I can do that. But since the packet size is essentially unbounded, and it can't be extended on the fly, I'll keep the splitting code.

I never copy the packet data. The AVPacket data is refcounted, and I release it when the last MMAL buffer is returned. There is one memcpy for the case an AVPacket is not refcounted (the user-visible FFmpeg API allows this, but hopefully it doesn't happen often). But the second memcpy in this function is just a bug. It memcpys using src==dst. It's a leftover from earlier and I fixed it.

>(I don't understand the if (avpkt && avpkt->data && avpkt->size) clause - is that the normal path for data?)

Similar as like MMAL, FFmpeg requires the user to send flush packets with size==0, but the other checks are redundant and a leftover from earlier.

Anyway, I still don't think the complex logic is avoidable, unless packets that are larger than expected are rejected.

One thing I'm not handling at all yet are format changes, especially resolution changes. I really have no idea how this is synchronized. One problem is that the API user still can have references to output buffers, so the pool can't just be destroyed. I refcount the pool now, which might make it easier.

Another question: does the port_disable/port_flush combination guarantee that all buffers have been returned to the user, including waiting for the port callbacks to finish? Currently I'm waiting for the buffers using a lock & condition variable.

6by9, 2015-03-20:
Your project, do what you like.

flush is a non-blocking request to return all buffers as soon as possible.

port_disable will block until the callback has been made for all buffers.

All the source for the MMAL core framework is in https://github.com/raspberrypi/userland/blob/master/interface/mmal/core and likewise for the ARM side of the VCHIQ comms (../vc/mmal_vc_api.c)

https://github.com/raspberrypi/userland/blob/master/interface/mmal/core/mmal_port.c#L696 mmal_port_disable_internal
```
/* Wait for all the buffers to have come back from the component */
LOG_DEBUG("%s waiting for %i buffers left in transit", port->name, (int)IN_TRANSIT_COUNT(port));
IN_TRANSIT_WAIT(port);
LOG_DEBUG("%s has no buffers left in transit", port->name);
```

wm4, 2015-03-20:
>port_disable will block until the callback has been made for all buffers.

I know, but port_flush is still documented as being asynchronous. I assume this asynchronous flush behavior is not even user-visible, then. I removed my own code for waiting for the buffers, and could simpliy it a lot.

The MMAL source is unfortunately not that useful, because all the interesting parts are on the videocore and closed source.

I pushed my updated code. It's much cleaner and simpler. I dropped the mutex too. This is probably what I will send to FFmpeg.

Thanks for your help, it made my life so much easier!

The core question of this issue can probably be answered with: there's no way, but the max. buffering is 16 frames.

closed 2015-03-20 by wm4