raspberrypi/userland #231: MMAL decoder doesn't error on unsupported h264 profiles

wm4, 2015-03-19:
Feeding a h264 Hi10P file to MMAL "works", but returns incorrect results. Most likely it just doesn't support this profile.

Why does MMAL not raise any kind of error when this profile is encountered? Is there a way to ask the decoder which profiles are supported? I found ``MMAL_VIDEO_PROFILE_T``, but I have no idea what to do with it.

popcornmix, 2015-03-19:
Hi10P is not supported. The basic rule is we support BluRay specs, and nothing higher.

It's debatable whether refusing to play these streams, or doing a "best effort" decode is more useful to the user. In general with Hi10P most macroblocks don't actually utilise the 10-bit encoding, so we can decode much of the file just fine. E.g.
https://www.youtube.com/watch?v=sNpn54GXm8o
https://www.youtube.com/watch?v=lFn_wj8kzCg

Any alternative (like a fallback to software decode) is likely to be far less useful for Hi10P.

wm4, 2015-03-19:
>Any alternative (like a fallback to software decode) is likely to be far less useful for Hi10P.

Well yes, but at least the API user should be able to make an informed decision about it. Sure, the set of supported profiles could be hardcoded, but that's not really ideal either.

JamesH65, 2015-03-19:
There is a MMAL function to return all the formats supported - use
 MMAL_PARAMETER_VIDEO_PROFILE_T  in a mmal_port_parameter_get() call.
Doxygen comments in code explain how.

wm4, 2015-03-19:
Thanks, I completely missed it. It was even under the MMAL_VIDEO_PROFILE_T enum.

closed 2015-03-19 by wm4