raspberrypi/userland #232: Excessive MMAL renderers cause display output to fail

waveform80, 2015-03-22:
A user of the picamera library reported an odd symptom when using a set of rather large image overlays: his display went completely black. When reproducing the issue I discovered that using up too much memory with MMAL renderers (which picamera uses to implement overlays) causes the display to attempt to switch to some sort of invalid mode (my HDMIPi monitor goes red and reports "out of range").

The following minimal script should reproduce the issue on a Pi with a default GPU memory split of 128Mb:

```python
import picamera
from PIL import Image
from time import sleep

# Create a blank image representing a 1920x1080 image in RGB format, padded out
# to 1920x1088 (to fit 32x16 blocks)
img = Image.new('RGB', (1920, 1088))

with picamera.PiCamera() as camera:
    camera.add_overlay(img.tostring(), size=(1920, 1080))
    camera.add_overlay(img.tostring(), size=(1920, 1080))
    camera.add_overlay(img.tostring(), size=(1920, 1080))
    sleep(5)
```

The script requires the python-picamera and python-imaging packages to be installed under Raspbian. A warning to anyone wanting to run the script: the display doesn't recover when the script ends (at least mine doesn't) so you'll probably want an SSH session into the Pi to safely reboot it afterwards.

I've tried running the script with various GPU memory splits all the way up to 256Mb but it seems to produce the same result in each case which makes me wonder whether this really is a lack of memory or something else? Anyway it would probably be preferable to have MMAL report some sort of error (ENOMEM?) in this event rather than disabling the display.

Original issue: waveform80/picamera#197

popcornmix, 2015-03-23:
That is expected behaviour. By default we composite all the dispmanx layers "on the fly". This means there is a real time constraint where for each horizontal line of HDMI output, we need to fetch all the source layers, resize/convert/blend and output the final pixels.

If you just keep adding layers, you hit the point where this breaks down.
Overclocking (expecially core_freq and sdram_freq) will help.

The real solution is to add dispmanx_offline=1 to config.txt. That no longer renders "on the fly" but uses an off screen buffer. It requires more gpu memory, and may reduce the update rate, but should avoid the issue you are seeing.

waveform80, 2015-03-23:
Interesting - thanks for the info. I'll add the dispmanx_offline info to the docs for picamera 1.10. Out of curiosity, is there a formula for calculating when one is about to exceed the time constraint? I'm mostly wondering if there's any way I could get picamera to throw an exception rather than disable the user's monitor?

popcornmix, 2015-03-23:
There is no certain way of predicting this beforehand. The busyness of axi bus and sdram will contribute to the failure.
It may be possible to poll (e.g. from a vcgencmd) if a display underrun has occurred.

waveform80, 2015-03-23:
Okay, no prob - I figured it would either be impossible or in the "very difficult" category, otherwise it would've already been implemented. I'll add some warnings to the overlay docs regarding the behaviour. Thanks!

closed 2015-03-23 by waveform80