raspberrypi/userland #235: Strange things happening on MMAL decoder format changes

wm4, 2015-03-25:
I tried implementing mid-stream format changes with the MMAL API. Since the documentation is scarce, I worked with the following assumption of how it should work: when receiving the new output format, I disable the output port, set the new format, commit it, allocate a new output pool, enable the port again, and done. The old output pool is deallocated while the output port is disabled (or a bit later if the libavcodec API user still has references to output frames). It seems obvious to me that the output pool needs to be recreated, because the size of the YUV image data change with a different video size.

This kind of works, except there is some memory corruption going on, which occasionally makes it crash. Apparently this is due to double-frees. It's almost as if the MMAL API itself messes with the pool, or has hidden references to it. So how does it work?

If anyone wants to take a look, I've pushed my code here: https://github.com/libav/libav/compare/master...wm4:rpi

wm4, 2015-03-26:
I made some progress. My changes added some other bug, which I had to fix first. Now I have working format changes, except for one incredibly weird issue with the MMAL API.

MMAL seems to return weird buffers on the output port right after format change. These buffers have the ``length`` field set to 0. How many of them are returned until normal buffers are returned seems to depend on the timing (I've seen 1 to 11). And worst of all, some of these buffers crash when passed to ``mmal_buffer_header_release()``. This happens because some buffers have _all_ fields set to 0, even ``priv``, which will make the buffer release function crash.

This workaround in addition to skipping output frames seems to work:

```C
            if (buffer->priv)
                mmal_buffer_header_release(buffer);
```

I'm absolutely baffled. This must be a bug somewhere else. Any hints?

valgrind doesn't show up any invalid reads or writes, but maybe it's just too slow to trigger the timing at all.

(I've updated the code on https://github.com/libav/libav/compare/master...wm4:rpi too.)

wm4, 2015-03-26:
PS: that MMAL_BUFFER_HEADER_T.priv is not set happens only when 1 buffer without data is returned from the output port. Only in cases in which priv is set there can be multiple buffers. I guess the latter case is normal, and this process merely returns the unused output buffers which were queued before the format change, and which now have the wrong size.

So the question is really why the API sometimes returns an invalid frame with all fields set to 0?

wm4, 2015-03-26:
Actually, I think I've found it. I didn't clear ``queue_decoded_frames`` on format change, so depending on the timing I was feeding already free'd buffers to the decoder. My other assumptions were apparently correct; I only had to drain the queue of decoded frame after disabling the output port.

(The MMAL API would actually be relatively nice, if not for this insane manual buffer/pool management...)

closed 2015-03-26 by wm4

6by9, 2015-03-28:
If you think MMAL is bad, you should try OpenMAX IL! 
MMAL was written to remove some of the insanity.

wm4, 2015-03-28:
>If you think MMAL is bad, you should try OpenMAX IL! 

No doubt - I was glad to find there's a simpler API on rpi!

MMAL could for example be improved by taking away pool management from the user.