raspberrypi/userland #236: Remove RaspiCamDocs.odt ?

lurch, 2015-03-27:
Now that all the [documentation](http://www.raspberrypi.org/documentation/) is hosted in it's own [repo](https://github.com/raspberrypi/documentation/), `host_applications/linux/apps/raspicam/RaspiCamDocs.odt` in this userland repo is becoming more and more out of date, which is especially evident when you compare the histories of [raspberrypi/documentation/raspbian/applications/camera.md](https://github.com/raspberrypi/documentation/commits/master/raspbian/applications/camera.md) and [raspberrypi/userland/host_applications/linux/apps/raspicam/RaspiCamDocs.odt](https://github.com/raspberrypi/userland/commits/master/host_applications/linux/apps/raspicam/RaspiCamDocs.odt).

It would probably make sense to just remove `host_applications/linux/apps/raspicam/RaspiCamDocs.odt` and simply treat http://www.raspberrypi.org/documentation/raspbian/applications/camera.md as the definitive camera-apps documentation? (Or, in case any third-party sites are linking directly to RaspiCamDocs.odt, another option would be to simply delete its current contents, and change it to say something like "The current documentation for the camera applications can be found at http://www.raspberrypi.org/documentation/raspbian/applications/camera.md" ?)

JamesH65, 2015-03-27:
Agreed. Now we have the decent repo based docs the ODT is no longer
necessary. I think your recommendation to change the text to point to the
online docs is the best option.

popcornmix, 2015-03-29:
A PR would be good.

6by9, 2016-01-01:
Change merged which removes RaspiCamDocs.odt. 
I didn't add a reference to the official page - linking to any file in a github repo would be rather brave, so I don't see it as being an issue that the file has now gone.

6by9, 2016-01-01:
Please close unless you think there is anything further to be done.

closed 2016-01-01 by lurch