raspberrypi/userland #238: mmal_port_parameter_get()/set() not idempotent for IntraRefresh.

thaytan, 2015-04-20:
Hi,

I was investigating https://github.com/thaytan/gst-rpicamsrc/issues/22 in gst-rpicamsrc (which is Raspivid turned into a GStreamer source element), in which using intra-refresh modes was causing an error.

Eventually, I figured out that mmal_port_parameter_set() with the MMAL_PARAMETER_VIDEO_INTRA_REFRESH_T causes problems, even though the struct is intialised with mmal_port_parameter_get() as in raspivid.

It looks like the air_ref, cir_mbs and pir_mbs fields are not populated by the call to mmal_port_parameter_get(), so they end up with whatever is on the stack. I guess in raspivid that's happening to always be 0, or some other valid value. memset'ing the struct to 0 before mmal_port_parameter_get() makes it always work for me, but it seems like the fields should be initted by the call.

popcornmix, 2015-04-21:
Okay, the real issue is that reading that parameter is not supported and so should return an error. RaspiVid doesn't check for an error.

I've added some untested code that hopefully supports reading the intra refresh parameter. Can you test it: https://dl.dropboxusercontent.com/u/3669512/temp/firmware_intra.zip

thaytan, 2015-04-22:
That firmware still rejects the mmal_port_parameter_get() for me:

0:00:02.620901194  3808  0x132eac0 ERROR              rpicamsrc RaspiCapture.c:1458:create_encoder_component: mmal_port_parameter_get() failed for MMAL_PARAMETER_VIDEO_INTRA_REFRESH
0:00:02.622496272  3808  0x132eac0 ERROR              rpicamsrc RaspiCamControl.c:767:mmal_status_to_int: Argument is invalid

RPi1, in case it matters.

popcornmix, 2015-04-22:
Okay, I've tested now. Please try: https://dl.dropboxusercontent.com/u/3669512/temp/firmware_intra.zip

thaytan, 2015-04-22:
Seems to work - the _get() call succeeds, and the various param setting vars get zeroed out.

popcornmix, 2015-04-22:
Okay will be in next firmware update.

thaytan, 2015-04-22:
Pull request that'll make raspivid warn and zero out the params, in case people happen to run newer userland against older firmware.

PR #239 (2015-04-23): Fix intra-refresh port parameter setting.
check mmal_port_parameter_get() succeeds when getting
the default intra refresh values. It fails on older firmware.

Fixes #238
commit 41b2b291 (2015-04-23): Fix intra-refresh port parameter setting.

check mmal_port_parameter_get() succeeds when getting
the default intra refresh values. It fails on older firmware.

Fixes #238
closed 2015-04-23 by popcornmix