raspberrypi/userland #242: Please update userland code.

asavah, 2015-05-25:
This repo is outdated after recent commits to https://github.com/raspberrypi/firmware.

Current userland code won't allow building of popcornmix/xbmc newclock4 branch with ffmpeg hevc patches https://github.com/popcornmix/xbmc/commit/98b12a72a249bd92c6b49b778418ee77868cd40a

Error: libavcodec/libavcodec.so: undefined reference to `vcsm_clean_invalid'

Output of nm from the relevant library built from current sources:
nm libvcsm.so | grep vcsm_clean_invalid
(nothing)

Output of nm from the relevant library from raspberrypi/firmware:
nm libvcsm.so | grep vcsm_clean_invalid
00003024 T vcsm_clean_invalid

Thanks.

popcornmix, 2015-05-25:
Sorry, missed that. Try now.

asavah, 2015-05-25:
TYVM!
Builds perfectly fine now.

closed 2015-05-25 by asavah