raspberrypi/userland #244: raspivid --help output is printed to stderr instead of stdout

derekenos, 2015-05-31:
This prevents pagers like `less` and `more` from working as expected when the height of the content exceeds that of the terminal window.

Example:

`raspivid --help | less` displays only the last page of content:
[image]

Redirecting `stderr` to `stdout` allows `less` to work as expected, `raspivid --help 2>&1 | less`:
[image]

6by9, 2015-06-01:
Feel free to sort out a patch and make a pull request.
You are right that all the fprintf lines are set to stderr, when stdout would probably be more useful.

Likewise I believe all the vcos_log_xxx lines are going to stderr. A quick tweak to https://github.com/raspberrypi/userland/blob/master/interface/vcos/pthreads/vcos_pthreads.c#L334 is likely to fix that, although that affects all users of vcos logging on the ARM, which includes the MMAL IPC framework and the likes.

popcornmix, 2015-06-02:
Be aware that raspivid supports output of encoded data to stdout and so you need to be careful not to mix logging output with that.

JamesH65, 2015-06-02:
Following on from popcornmix, all output was pushed to stderr in order to
enable the encoded output to stdout. However, I see no reason off hand why
the help cannot go to stdout, since you won't be using them at the same
time as encoded output.

6by9, 2016-01-01:
Change merged that sends the help text to stdout instead of stderr.
Please retest and close, unless you think there is something further to do.

6by9, 2016-01-16:
@popcornmix Do you want to close this? The issue is fixed, and the original poster seems to have gone away.

closed 2016-01-16 by popcornmix

derekenos, 2016-01-16:
Sorry, I've been meaning to respond. Thanks for the fix!