raspberrypi/userland #245: Missing .pc files

XECDesign, 2015-06-08:
It would be nice to have bcm_host.pc, egl.pc and glesv2.pc included. It would reduce the modifications needed to compile software for the pi.

leio, 2015-06-08:
For these to be really useful, they should be installed in the standard location - $libdir/pkgconfig (probably /usr/lib/pkgconfig most of the time) - instead of where it would be installed usually per PREFIX (like PR#110 would) under /opt/vc/*
Which leads to the question of why all this is in /opt anyways, if open source. If it installed in a normal prefix, pkgconfig files would go to the right place automatically.
If they were to go to /usr/lib/pkgconfig, then the names are unsuitable - egl and glesv2 in particular, as mesa uses these too. So if the names stay and userland stays in /opt too, /opt/vc is a good place for .pc for them to not clash with other files, and we downstream can play around with PKG_CONFIG_PATH and co where needed.

I believe until the rest stays in /opt/vc, pkgconfig files as coming from upstream should be there too, and then I can decide if I want to move them around or point everything to them via extra PKG_CONFIG_PATH kind of things, but then the same name with mesa can still be complicating matters. Or maybe install everything with /usr PREFIX instead of /opt/vc.

modwizcode, 2015-12-14:
Updates on this? This is seriously frustrating and the number of accidental installs of mesa based libraries (which then can't easily be fully removed to fix their mess) caused by the lack of pkg-config files is huge. This isn't even a difficult fix, it's basically just 3 stub files.

popcornmix, 2015-12-14:
Pull requests are welcome.

jgueytat, 2016-05-23:
Ok I'll take care of the push request.

jgueytat, 2016-05-23:
Here are the current *.pc files I'm using. It comes from the meta-raspberrypi repository for the Yocto project.
Could you please double check those? I will then integrate them and make the pull request.

Best Regards,

**bcm_host.pc**
$ cat bcm_host.pc
prefix=/usr
exec_prefix=${prefix}
libdir=${exec_prefix}/lib
includedir=${prefix}/include

Name: bcm_host
Description: Broadcom VideoCore host API library
Version: 1
Libs: -L${libdir} -lbcm_host -lvcos -lvchiq_arm -pthread
Cflags: -I${includedir} -I${includedir}/interface/vmcs_host/linux -I${includedir}/interface/vcos/pthreads -DUSE_VCHIQ_ARM

**egl.pc**
$ cat egl.pc
prefix=/usr
exec_prefix=${prefix}
libdir=${exec_prefix}/lib
includedir=${prefix}/include

Name: EGL
Description: Fake EGL package for RPi
Version: 10
Requires: bcm_host
Libs: -L${libdir} -lEGL -lGLESv2 -lbcm_host -lvchostif
Cflags: -I${includedir} -I${includedir}/interface/vmcs_host/linux \
-I${includedir}/interface/vcos/pthreads

**glesv2.pc**
$ cat glesv2.pc
prefix=/usr
exec_prefix=${prefix}
libdir=${exec_prefix}/lib
includedir=${prefix}/include

Name: GLESv2
Description: Fake GL ES 2 package for RPi
Version: 10
Requires: bcm_host
Libs: -L${libdir} -lGLESv2
Cflags: -I${includedir}

**vg.pc**
$ cat vg.pc
prefix=/usr
exec_prefix=${prefix}
libdir=${exec_prefix}/lib
includedir=${prefix}/include

Name: OpenVG
Description: Fake OpenVG package for RPi
Version: 10
Requires: bcm_host
Libs: -L${libdir} -lOpenVG
Cflags: -I${includedir}

AndrewFromMelbourne, 2016-05-23:
Thanks for the update.

jgueytat, 2016-05-23:
The code is actually used in Yocto thanks to kraj. :)
https://github.com/kraj/userland/commit/e03358e0b2887afc733df1cdf2ea9aa1aa3573ef

jgueytat, 2016-05-23:
@popcornmix the pull request has already been done.
I think you will find  a lot of nice improvements in there.
https://github.com/raspberrypi/userland/pull/290

Best Regards,
Hope to see good news soon,

popcornmix, 2016-05-31:
Latest firmware includes the .pc files from https://github.com/kraj/userland/commit/e03358e0b2887afc733df1cdf2ea9aa1aa3573ef
See: https://github.com/raspberrypi/firmware/tree/master/opt/vc/lib/pkgconfig
@XECDesign is this as desired?

XECDesign, 2016-05-31:
They're there and they're usable with the appropriate PKG_CONFIG_PATH variable, so it looks good to me.

closed 2016-05-31 by popcornmix

sirdel, 2016-06-19:
On a clean raspbian install pkg-config checks these locations:
`pkg-config --variable pc_path pkg-config /usr/local/lib/arm-linux-gnueabihf/pkgconfig:/usr/local/lib/pkgconfig:/usr/local/share/pkgconfig:/usr/lib/arm-linux-gnueabihf/pkgconfig:/usr/lib/pkgconfig:/usr/share/pkgconfig`

What is the appropriate PKG_CONFIG_PATH ??

popcornmix, 2016-06-20:
I imagine it's ``/opt/vc/lib/pkgconfig/`` as that is where the nre ``.pc`` files are.