raspberrypi/userland #246: Cannot glMapBuffer with GL_STRAM_DRAW buffer data

b-pass, 2015-06-14:
In raspberrypi/userland/interface/khronos/glxx/glxx_client.c
Line 385 & 397
... If glBufferData() is called with GL_STREAM_DRAW then the buffer's cached size is set to zero.

In raspberrypi/userland/interface/khronos/ext/gl_oes_map_buffer.c
Line 66 & Line 94
... If the cached size is zero, then the GL_INVALID_OPERATION is raised and the buffer is not mapped.

The net effect being that you can't call glMapBufferOES on a buffer that was set up using GL_STREAM_DRAW.  I think it ought to work fine though, and it seems like a small bug on glxx_client.c line 385.

JamesH65, 2018-07-02:
@timg236 Anything you can help with? It's quite an old issue...

timg236, 2018-07-02:
I think @b-pass is right. This looks like a bug in the userland glxx_client.c. I hacked hello_triangle.c to map, modify and unmap the buffer and was able to reproduce the failure. The trivial fix of accepting GL_STREAM_DRAW in glxx_client.c line 385 resolved the issue.

The legacy driver is deprecated but I think the fix is simple  enough that we should just fix it.

It would be useful to know if there is test-case for this.

JamesH65, 2018-07-02:
@timg236 Can you push the change as a PR? No idea about a test case.

PR #471 (2018-07-02): glxx_client: Fix glMapBufferOES with GL_STREAM_DRAW
glMapBufferOES fails if the usage hint passed to glBufferData was
GL_STREAM_DRAW because glBufferData does not initialise the cached copy
of the buffer size.

The usage hint doesn't make any functional difference with these
functions and GL_STREAM_DRAW was added in GLES 2.0 so this looks like
an old check which was never updated.

Fixes #246
commit 8ccdc868 (2018-07-02): glxx_client: Fix glMapBufferOES with GL_STREAM_DRAW

glMapBufferOES fails if the usage hint passed to glBufferData was
GL_STREAM_DRAW because glBufferData does not initialise the cached copy
of the buffer size.

The usage hint doesn't make any functional difference with these
functions and GL_STREAM_DRAW was added in GLES 2.0 so this looks like
an old check which was never updated.

Fixes #246
closed 2018-07-02 by JamesH65