raspberrypi/userland #249: Some Khronos headers seem outdated

LongChair, 2015-07-05:
While trying to rebuild some recent apps on RPI2, i figured that some Khronos headers are lacking some definitions from the official ones.

Here is the reference one https://www.khronos.org/registry/egl/api/KHR/khrplatform.h

The RPI one is at least missing the folowing definitions at the end of the file :

```
/*
 * Dummy value used to pad enum types to 32 bits.
 */
#ifndef KHRONOS_MAX_ENUM
#define KHRONOS_MAX_ENUM 0x7FFFFFFF
#endif

/*
 * Enumerated boolean type
 *
 * Values other than zero should be considered to be true.  Therefore
 * comparisons should not be made against KHRONOS_TRUE.
 */
typedef enum {
    KHRONOS_FALSE = 0,
    KHRONOS_TRUE  = 1,
    KHRONOS_BOOLEAN_ENUM_FORCE_SIZE = KHRONOS_MAX_ENUM
} khronos_boolean_enum_t;
```

Any chance we could have those synced ?

Ruffio, 2016-09-04:
@LongChair this issue looks like it has been fixed but forgot to be closed?

JamesH65, 2017-12-05:
Indeed, I believe this has been fixed, so closing.

closed 2017-12-05 by JamesH65