raspberrypi/userland #253: hello_video freezes

fredericofs, 2015-08-29:
My hello_video plays small h264 movies ok.
But freezes along the way when playing longer files.
This example, always freezes for me near the end:
https://www.dropbox.com/s/xlyaflmmij4unms/2menu.h264?dl=0

fredericofs, 2015-09-23:
fixed when removing timeout, I am sending a pull request

closed 2015-09-23 by fredericofs

tvjon, 2019-03-04:
[hello_audio.bin.zip](https://github.com/raspberrypi/userland/files/2928520/hello_audio.bin.zip)

Attached is a zip of compiled hello_audio.bin

When it runs, there's an anoying buzz present in the audio it produces.

Typical audio files played through the audio jack sound fine.

$ vcgencmd version
Feb 12 2019 22:14:33 
Copyright (c) 2012 Broadcom
version 8eff5e4023657a8b3b59e1f90dc966f62d74908c (clean) (release) (start_db)

$ uname -a
Linux sa1 4.19.20-v7+ #1201 SMP Tue Feb 12 22:10:44 GMT 2019 armv7l GNU/Linux