raspberrypi/userland #255: "undefined reference to `mmal_"* - compilation fails (with gcc-4.8.5?)

srcshelton, 2015-09-04:
Commit b864a841e5a459a66a890c22b3a34127cd226238 built correctly with gcc-4.8.4 - but now this and all commits since are failing with gcc-4.8.5.

The build process starts:

```make
cmake --no-warn-unused-cli -C /var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999_build/gentoo_common_config.cmake -G Unix Makefiles -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Gentoo -DCMAKE_INSTALL_DO_STRIP=OFF -DCMAKE_USER_MAKE_RULES_OVERRIDE=/var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999_build/gentoo_rules.cmake -DCMAKE_TOOLCHAIN_FILE=/var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999_build/gentoo_toolchain.cmake  /var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999
Not searching for unused variables given on the command line.
loading initial cache file /var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999_build/gentoo_common_config.cmake
-- The C compiler identification is GNU 4.8.5
-- The CXX compiler identification is GNU 4.8.5
-- Check for working C compiler: /usr/bin/armv7a-hardfloat-linux-gnueabi-gcc
-- Check for working C compiler: /usr/bin/armv7a-hardfloat-linux-gnueabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/armv7a-hardfloat-linux-gnueabi-g++
-- Check for working CXX compiler: /usr/bin/armv7a-hardfloat-linux-gnueabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for execinfo.h
-- Looking for execinfo.h - found
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/armv7a-hardfloat-linux-gnueabi-gcc
CMake Warning at interface/usbdk/CMakeLists.txt:2 (message):
  usbdk: using stubbed out hostreq, HDMI buttons and gestures code

-- <<< Gentoo configuration >>>
Build type      Gentoo
Install path    /opt/vc
Compiler flags:
C               -Os -pipe -mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -funsafe-math-optimizations  -Wno-multichar -Wall -Wno-unused-but-set-variable -fPIC
C++             -Os -pipe -mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -funsafe-math-optimizations 
Linker flags:
Executable       -Wl,-O1 -Wl,--as-needed
Module           -Wl,-O1 -Wl,--as-needed
Shared           -Wl,-O1 -Wl,--as-needed

-- Configuring done
-- Generating done
-- Build files have been written to: /var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999_build
>>> Source configured.
>>> Compiling source in /var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999 ...
>>> Working in BUILD_DIR: "/var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999_build"
make -j5 VERBOSE=1
/usr/bin/cmake -H/var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999 -B/var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999_build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999_build/CMakeFiles /var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999_build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999_build'
make -f interface/vcos/pthreads/CMakeFiles/vcos.dir/build.make interface/vcos/pthreads/CMakeFiles/vcos.dir/depend
make -f interface/vchiq_arm/CMakeFiles/vchiq_arm.dir/build.make interface/vchiq_arm/CMakeFiles/vchiq_arm.dir/depend
make -f interface/vmcs_host/linux/vcfiled/CMakeFiles/vcfiled_check.dir/build.make interface/vmcs_host/linux/vcfiled/CMakeFiles/vcfiled_check.dir/depend
make -f interface/khronos/CMakeFiles/khrn_client.dir/build.make interface/khronos/CMakeFiles/khrn_client.dir/depend
make -f interface/khronos/CMakeFiles/EGL_static.dir/build.make interface/khronos/CMakeFiles/EGL_static.dir/depend
make[2]: Entering directory '/var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999_build'
```

... and the final error is:

```make
make -f interface/khronos/CMakeFiles/WFC.dir/build.make interface/khronos/CMakeFiles/WFC.dir/build
make[2]: Entering directory '/var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999_build'
/usr/bin/cmake -E cmake_progress_report /var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999_build/CMakeFiles 20
[ 89%] Building C object interface/khronos/CMakeFiles/WFC.dir/wf/wfc_client_stream.c.o
cd /var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999_build/interface/khronos && /usr/bin/armv7a-hardfloat-linux-gnueabi-gcc  -DEGL_SERVER_DISPMANX -DHAVE_CMAKE_CONFIG -DHAVE_VMCS_CONFIG -DOMX_SKIP64BIT -DTV_SUPPORTED_MODE_NO_DEPRECATED -DUSE_VCHIQ_ARM -DVCHI_BULK_ALIGN=1 -DVCHI_BULK_GRANULARITY=1 -DWFC_EXPORTS -D_FILE_OFFSET_BITS=64 -D_HAVE_SBRK -D_LARGEFILE64_SOURCE -D_LARGEFILE_SOURCE -D_REENTRANT -D__VIDEOCORE4__  -DNDEBUG -Os -pipe -mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -funsafe-math-optimizations  -Wno-multichar -Wall -Wno-unused-but-set-variable -fPIC -fPIC -I/var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999/build/inc -I/var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999/host_applications/framework -I/var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999 -I/var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999/interface/vcos/pthreads -I/var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999/interface/vmcs_host/linux -I/var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999/interface/vmcs_host -I/var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999/interface/vmcs_host/khronos -I/var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999/interface/khronos/include -I/var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999_build -I/var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999/interface/vchiq_arm -I/var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999/host_support/include    -o CMakeFiles/WFC.dir/wf/wfc_client_stream.c.o -c /var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999/interface/khronos/wf/wfc_client_stream.c
Linking C executable /var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999/build/bin/mmal_example_basic_1
cd /var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999_build/interface/mmal/test && /usr/bin/cmake -E cmake_link_script CMakeFiles/mmal_example_basic_1.dir/link.txt --verbose=1
/usr/bin/armv7a-hardfloat-linux-gnueabi-gcc  -Os -pipe -mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -funsafe-math-optimizations  -Wno-multichar -Wall -Wno-unused-but-set-variable -fPIC   -Wl,-O1 -Wl,--as-needed CMakeFiles/mmal_example_basic_1.dir/examples/example_basic_1.c.o  -o /var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999/build/bin/mmal_example_basic_1 -rdynamic -Wl,--whole-archive /var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999/build/lib/libmmal_components.so -Wl,--no-whole-archive /var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberr [... cut ...]
make[2]: *** [/var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999/build/bin/mmal_example_basic_1] Error 1
make[2]: Leaving directory '/var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999_build'
CMakeFiles/Makefile2:1384: recipe for target 'interface/mmal/test/CMakeFiles/mmal_example_basic_1.dir/all' failed
make[1]: *** [interface/mmal/test/CMakeFiles/mmal_example_basic_1.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

srcshelton, 2015-09-04:
FUll build log available at: https://gist.githubusercontent.com/srcshelton/9a1f8a3618e454d456a0/raw/b591119f5a2d7c9c2fb62aeed9e3578ac93a7ca8/raspberrypi-userland-9999.log

popcornmix, 2015-09-05:
Can you be more specific:

> Commit b864a84 built correctly with gcc-4.8.4 - but now this and all commits since are failing with gcc-4.8.5.

Are you saying that b864a84 is the first commit that fails, and preceding commit (ffdd891) works?
Considering the content of b864a84 that doesn't seem possible.

srcshelton, 2015-09-05:
No, sorry, I'm saying that I was using gcc-4.8.4, and with this commit b864a84 compiled and ran correctly.  Since b864a84, I've now upgraded to gcc-4.8.5 and now neither b864a84 nor any commit since works.  As b864a84 did work before, I've not tracked any further backwards.

ED6E0F17, 2015-09-07:
Nice overlay, I like what you did with the pthreads issue in  https://github.com/raspberrypi/firmware/issues/34

A direct pull of master builds fine on ChromiumOS, which is basically the same as Gentoo, but the same  issue was reported here:  http://lists.opensuse.org/opensuse-arm/2015-01/msg00041.html

popcornmix, 2015-09-07:
The userland repo is designed for raspbian and we can't offer much help on getting it working on other distributions. We are willing to accept patches that fix build issues on other distributions if they don't harm raspbian.
I suspect your issue is something trivial. The missing symbols (e.g. mmal_list_create) come from libmmal_util.so. 

Firstly confirm you have that symbol. E.g:
```
dc4@dc4-XPS13-9333:~/vc4$ arm-linux-gnueabihf-objdump  -t /home/dc4/rootfs/opt/vc/lib/libmmal_util.so | grep mmal_list_create
0000a840 g     F .text	0000007c              mmal_list_create
```
Assuming you do it's probably down to the order of the libraries on the build line. I assume this is the build line that is failing:
```
/usr/bin/armv7a-hardfloat-linux-gnueabi-gcc  -Os -pipe -mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -funsafe-math-optimizations  -Wno-multichar -Wall -Wno-unused-but-set-variable -fPIC   -Wl,-O1 -Wl,--as-needed CMakeFiles/mmal_example_basic_1.dir/examples/example_basic_1.c.o  -o /var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999/build/bin/mmal_example_basic_1 -rdynamic -Wl,--whole-archive /var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999/build/lib/libmmal_components.so -Wl,--no-whole-archive /var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999/build/lib/libmmal_core.so /var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999/build/lib/libmmal_util.so /var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999/build/lib/libcontainers.so /var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999/build/lib/libvcos.so -lpthread -ldl -lrt -Wl,-rpath,/var/.mem/cache/portage/build/portage/media-libs/raspberrypi-userland-9999/work/raspberrypi-userland-9999/build/lib 
```
Can you try adding ``-Wl,--start-group`` before the libs and ``-Wl,--end-group`` after the libs, and see if that fixes the build command.

srcshelton, 2015-09-10:
Ah - the problem in my case (although I'm not sure how this relates to SuSE) is that `-Wl,--as-needed`, which is in `LDFLAGS` by default, is breaking things in this instance: this class of problem is documented in general terms here: [Project:Quality_Assurance/As-needed](https://wiki.gentoo.org/wiki/Project:Quality_Assurance/As-needed#Failure_in_final_linking.2C_undefined_symbols)

Weird that this previously worked successfully with the same setting in place, though...

popcornmix, 2015-09-10:
Can you confirm that removing ``-Wl,--as-needed`` produces a working build for you?

srcshelton, 2015-09-11:
Yes, it's fine with `-Wl,--as-needed` removed - [raspberrypi-userland-9999.ebuild](https://github.com/srcshelton/gentoo-ebuilds/blob/master/media-libs/raspberrypi-userland/raspberrypi-userland-9999.ebuild) is now a working ebuild.

Ruffio, 2016-09-04:
@popcornmix this issue should be closed as it has been resolved

ED6E0F17, 2016-09-06:
Nothing has been  resolved - but it is not an issue with the supported compiler.

It remains an issue on Gentoo, Android and others, that can be fixed with  https://github.com/raspberrypi/userland/pull/179 , but that may not be the best solution.

6by9, 2016-09-06:
It can be closed as a duplicate of #176, #303, and potentially others.

The requirements of any updated mechanism have been stated at https://github.com/raspberrypi/userland/issues/303#issuecomment-204394592, but those who are complaining don't appear to be prepared to put the legwork in to fix, even though it is their change to the build settings that is causing the breakage. Ho hum.

popcornmix, 2016-09-06:
Closing as duplicate of #178

closed 2016-09-06 by popcornmix