raspberrypi/userland #256: buildme error on Arch Linux Arm

eredacokmerke, 2015-09-05:
I tried to compile userland on Pi2 with Arch Linux Arm.After I type ./buildme I get 
```
./buildme: line 3: arch: command not found
./buildme: line 3: [: armv6l: unary operator expected
./buildme: line 3: arch: command not found
./buildme: line 3: [: armv7l: unary operator expected
~/Desktop/userland-master/build/arm-linux/release ~/Desktop/userland-master
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
```
error. I typed buildme commands manually
```
mkdir -p build/raspberry/release
pushd build/raspberry/release
cmake -DCMAKE_BUILD_TYPE=Release ../../..
make -j4
```
```make -j4``` returns error
```
cc1: all warnings being treated as errors
interface/vcos/pthreads/CMakeFiles/vcos.dir/build.make:62: recipe for target 'interface/vcos/pthreads/CMakeFiles/vcos.dir/vcos_pthreads.c.o' failed
make[2]: *** [interface/vcos/pthreads/CMakeFiles/vcos.dir/vcos_pthreads.c.o] Error 1
cc1: all warnings being treated as errors
interface/vcos/pthreads/CMakeFiles/vcos.dir/build.make:134: recipe for target 'interface/vcos/pthreads/CMakeFiles/vcos.dir/__/generic/vcos_generic_event_flags.c.o' failed
make[2]: *** [interface/vcos/pthreads/CMakeFiles/vcos.dir/__/generic/vcos_generic_event_flags.c.o] Error 1
CMakeFiles/Makefile2:105: recipe for target 'interface/vcos/pthreads/CMakeFiles/vcos.dir/all' failed
make[1]: *** [interface/vcos/pthreads/CMakeFiles/vcos.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

My system is Arch Linux Arm . ```Linux alarmpi 4.1.6-2-ARCH #1 SMP PREEMPT Sat Aug 22 12:18:37 MDT 2015 armv7l GNU/Linux```

clivem, 2015-09-21:
This "smells" like using gcc 5.x to compile it. Try adding "-fgnu89-inline" to your CFLAGS.

Ruffio, 2016-09-04:
@eredacokmerke did @clivem suggestion Sølve your issue?

6by9, 2016-09-05:
At a cursory glance, looks like a duplicate of #220, resolved in January 2016.

Ruffio, 2016-12-30:
This issue should be closed due to lact of activity and it looks like and Xref has been solved in Jan 2016. Last activity on this issue was on 21 Sep 2015, and I pinged on 4 Sep 2016 without any response...

closed 2016-12-30 by popcornmix