raspberrypi/userland #257: Debian Packaging

robertely, 2015-09-17:
Hey guys, Is there an official packaged version of this available anywhere? 

If not I'd like to help get that setup if possible. I want to use this in Debian Jessie, and while I can compile it each time I build Debian it should IMHO be in a package (one not named "userland".)

XECDesign, 2015-09-18:
Such a package does not currently exist.

TheSin-, 2016-03-01:
this is not true I build all this stuff in my repo for my image builder scripts

https://github.com/TheSin-/rpi-img-builder

I'm currently getting arm64 all setup to be added as well.  Feel free to use my public repo if you'd like.

clivem, 2016-03-01:
arm64, you say? Good luck with that! <evil laugh>

alexellis, 2016-06-09:
Is there a binary distribution that I can add to Raspbian? I'm using resin/rpi-raspbian as a Docker image and it doesn't contain raspistill?

Ruffio, 2016-12-30:
@robertely I think that this should be asked in Forum. Github is only for bugs. This should be closed...

closed 2017-12-05 by JamesH65