raspberrypi/userland #260: User buffer flags not preserved on input port

wm4, 2015-09-23:
Quoting the docs:
```
/** User flags - can be passed in and will get returned */
#define MMAL_BUFFER_HEADER_FLAG_USER0                  (1<<28)
#define MMAL_BUFFER_HEADER_FLAG_USER1                  (1<<29)
#define MMAL_BUFFER_HEADER_FLAG_USER2                  (1<<30)
#define MMAL_BUFFER_HEADER_FLAG_USER3                  (1<<31)
```
So I'd expect them not to vanish when setting those flags on a buffer, sending it to an input port, and then receiving the buffer again through the input port's callback. But the input port callback for the VC h264 decoder always returns buffers with flags=0. Is this a bug?

popcornmix, 2015-09-24:
No, the intention is they come back from the output port of video_decode.
I believe recycled buffers are always deliberately cleared.
I use these with with video_decode tunnelled to image_fx to distinguish pictures directly associated with a submitted input buffer from the additional deinterlaced frames.

e.g. I set MMAL_BUFFER_HEADER_FLAG_USER0 on each encoded frame, and get that flag back for the primary pictures, and no flags back from the extra deinterlaced pictures.

wm4, 2015-09-24:
That's a bit unexpected. I worked it around by putting the information into user_data. Feel free to close this if it's not considered a bug.

closed 2015-09-24 by popcornmix