raspberrypi/userland #269: Please keep this repo in sync with raspberrypi/firmware

asavah, 2015-11-15:
Include related changes are missing at least from
https://github.com/raspberrypi/firmware/commit/591b25a7749027ce095a49d0a515294491dd26fc 
and
https://github.com/raspberrypi/firmware/commit/d56bf185f0f9e1cc2d9090970a2db1f6ebe76bb2

popcornmix, 2015-11-16:
I've updated userland. Okay now?

asavah, 2015-11-16:
Thank you.
I think these are still missing: https://github.com/raspberrypi/firmware/blob/master/opt/vc/include/interface/vmcs_host/vc_vchi_audioserv_defs.h#L32
https://github.com/raspberrypi/firmware/blob/master/opt/vc/include/interface/vmcs_host/vc_vchi_audioserv_defs.h#L53
https://github.com/raspberrypi/firmware/blob/master/opt/vc/include/interface/vmcs_host/vc_vchi_audioserv_defs.h#L77
https://github.com/raspberrypi/firmware/blob/master/opt/vc/include/interface/vmcs_host/vc_vchi_audioserv_defs.h#L114
https://github.com/raspberrypi/firmware/blob/master/opt/vc/include/interface/vmcs_host/vc_vchi_audioserv_defs.h#L158

popcornmix, 2015-11-16:
Okay, added.

closed 2015-11-16 by asavah