raspberrypi/userland #270: EV? ISO? AwbGains? ShutterSpeed?

schabbeskugel, 2015-11-25:
Dear Reader,
I am not sure if all parameters of raspistill are working as intentionally.
As with Infrared lighting, I wanted to fixate awbgain, shutterspeed etc.
so always using -awb off -awbg 1,2 -ss 600000 etc.
but:
-ev does not show any effect (nor -10 nor +10)
-awb off seems to work for the preview window, but then there seem to be a second pass, where my settings are overwritten by some kind of automatism even with -awb off. strings -a test.jpg |head shows really different values.
-ISO does not work for mee, too, trying different values.
-ss 600000 does take approx. 3-4s before jpg output, with -t  1(ms!). As described above, there seem to be two passes...

Does anybody know something about this?
Kind regards, schabbeskugel

6by9, 2015-11-30:
Changing EV alters the AGC target. If you've specified shutter speed and ISO then you've disabled the automatic algorithm in favour of your manual settings. What do you expect it to do?

awb off with manually specified gains does work on the latest firmwares - what version are you running (```vcgencmd version```). Do NOT specify -st (for statistics pass) on the command line or your settings WILL be overwritten - that is deliberate.

ISO will fix the analogue and digital gains. What values are you trying?

If you ask for a 600ms exposure with -t 1, then it will take approx 4 frame periods, or 2.4s to capture. 
Two facts:
 - The system always completes any request. 
 - The first frame off almost all Bayer sensors is corrupted and has to be dropped.
The sensor will start in preview mode and get a request. That will take 600ms to create the corrupt frame, and a further 600ms to produce the preview frame. The sensor is then stopped and reprogrammed for the full 5MP mode, restarted, and will take 600ms to produce a corrupt frame, and a further 600ms to produce the frame that becomes the JPEG. Sorry, that's the way it is. 

Please provide EXACTLY the commands that you are using if you want further analysis.

6by9, 2015-12-23:
@schabbeskugel please respond or close the issue.

schabbeskugel, 2015-12-23:
Dear 6by9,

thanks for your quick response; I'll now have two cams (one in production, one for testing) and hope to completely respond tomorrow;

testing (pi 2) is Nov 18 2015 15:48:41 
Copyright (c) 2012 Broadcom
version 0c9af7cde38e223c95c3404f468db8fc5004495c (clean) (release)

"production" (where I did the first tests) (pi b) is Sep 23 2015 12:21:14 
Copyright (c) 2012 Broadcom
version c156d00b148c30a3ba28ec376c9c01e95a77d6d5 (clean) (release)

Kind regards,
schabbeskugel

Ruffio, 2016-12-30:
@schabbeskugel please respond or close the issue.

Serveurperso, 2017-09-19:
I can't set the - ISO parameters, it's always ignored even with - ex off

raspivid -fps 25 -w 720 -h 576 -ex off -ISO 800 -cd MJPEG -o /var/www/temp/video.mjpg

tested many combinations of - ex off/auto/night without success

How I can push the gain, I want to see the sensor noise!

Serveurperso, 2017-09-19:
(root|~) vcgencmd version
Sep 12 2017 15:25:20
Copyright (c) 2012 Broadcom
version 3fbb171d1950eb3789eb8e37ba11e01079a37239 (clean) (release)
(root|~)

all last available version on raspberry pi 3 (stretch)

6by9, 2017-10-10:
Closing due to inactivity. Please reopen if there is still an issue or questions to be answered.

closed 2017-10-10 by 6by9