raspberrypi/userland #271: Fails when building with a musl based toolchain

sergioprado, 2015-11-27:
When building with a musl based toolchain, the build fails with the following message:

---------------------------------------------------------------------------------
[ 96%] Building C object host_applications/linux/apps/raspicam/CMakeFiles/raspividyuv.dir/RaspiVidYUV.c.o
In file included from /home/sprado/workspace/build/buildroot/autobuild/rpi-userland/build/rpi-userland-8306b7c757f4b8e8d9148789eeb64c666d9244a5/host_applications/linux/apps/raspicam/RaspiVidYUV.c:55:0:
/home/sprado/workspace/build/buildroot/autobuild/rpi-userland/build/rpi-userland-8306b7c757f4b8e8d9148789eeb64c666d9244a5/host_applications/linux/apps/raspicam/RaspiVidYUV.c:109:14: error: conflicting type qualifiers for ‘stderr’
 extern FILE *stderr, *stdout;
              ^
/home/sprado/workspace/build/buildroot/autobuild/rpi-userland/host/usr/arm-buildroot-linux-musleabihf/sysroot/usr/include/stdio.h:57:20: note: previous declaration of ‘stderr’ was here
 extern FILE *const stderr;
                    ^
---------------------------------------------------------------------------------

That's because musl defines stderr and stdout as a const pointer:

extern FILE *const stdin;
extern FILE *const stdout;
extern FILE *const stderr;

And the application (RaspiVidYUV.c) don't:

extern FILE *stderr, *stdout;

Commenting this line solves this problem. But, is this the proper solution to the problem?

JamesH65, 2015-11-27:
Does that break the normal build? Not entirely sure why I added that line
in the first place as it should come in on the header.

sergioprado, 2015-11-27:
No, it doesn't. I've tested with both glibc and musl without this line and it works.

sergioprado, 2015-11-27:
I've just send a pull request to fix this problem.

https://github.com/raspberrypi/userland/pull/272

6by9, 2015-12-23:
PR is merged. Please close the issue if you're now happy.

closed 2015-12-23 by sergioprado