raspberrypi/userland #273: Low Resolution snapshot error

ixs0792, 2015-12-03:
There seems to be an error in the vc_dispmanx_resource_create, vc_dispmanx_rect_set, vc_dispmanx_snapshot, vc_dispmanx_resource_read_data command chain that causes some videos to produce incorrect results when using low width downscaling. 

I am able to recreate the error using [raspi2png] (https://github.com/AndrewFromMelbourne/raspi2png) to output 32x18 files. 
When playing a video via omxplayer that is "Video: h264 (Main)" encoded it works fine.
However, playing a video that is "Video: h264 (High)" encoded it produces a nearly black output.  

Switching to 64x36 output both videos work fine. 

This doesn't seem too critical (a 2:1 downscale is easy enough), but it drove me crazy for a few days trying to figure it out...

AndrewFromMelbourne, 2015-12-03:
This is the wrong place to report/discuss this issue. 

If it is an issue with the DispmanX interface, then it should be reported in the [firmware issues](https://github.com/raspberrypi/firmware/issues).

It could also be an issue with the way I am using the DispmanX API. It is undocumented, so there is every chance that I am not using it correctly. So it would better as a [raspi2png issue](https://github.com/AndrewFromMelbourne/raspi2png/issues).

ixs0792, 2015-12-03:
raspi2png doesn't seem to be the issue. I've recreated it using fbcp and my own code. All of them use the undocumented DispmanX API the same way. raspi2png is just the easiest way to recreate it (and is nice and easy to use - thanks!). 

So it seems like either the standard interpretation of these commands are incorrect, or there is an issue with the API.

closed 2015-12-03 by ixs0792

AndrewFromMelbourne, 2015-12-03:
Open another issue in the firmware issues. @popcornmix has been very helpful to me on similar issues in the past.