raspberrypi/userland #274: rastpstill with timelapse and datetime does not include Year

mfn, 2015-12-05:
The `--help` states:
`-dt, --datetime : Replace frame number in file name with DateTime (YearMonthDayHourMinSec)`

Using this command line, however, the generated files do not include the year:

`raspistill --width 1920 --height 1080 --encoding png --timelapse 10000 --datetime --timeout 0 --output image_%d.png`

Generated files:
```
pi@raspberrypi:~/tmp $ ls -l
total 680
-rw-r--r-- 1 pi pi 39075 Dec  5 23:15 image_1205231531.png
```
It should be `image_20151205231531.png`

My c-fu is limited, but the code in [RaspiStill.c](https://github.com/raspberrypi/userland/blob/daad93e972f2332555ca2dc7b5a27a89d52572d5/host_applications/linux/apps/raspicam/RaspiStill.c#L1803-L1819) looks like it simply doesn't take the year into account:
```C
                if (state.datetime)
                {
                   time_t rawtime;
                   struct tm *timeinfo;

                   time(&rawtime);
                   timeinfo = localtime(&rawtime);

                   frame = timeinfo->tm_mon+1;
                   frame *= 100;
                   frame += timeinfo->tm_mday;
                   frame *= 100;
                   frame += timeinfo->tm_hour;
                   frame *= 100;
                   frame += timeinfo->tm_min;
                   frame *= 100;
                   frame += timeinfo->tm_sec;
                }
```

6by9, 2015-12-09:
A signed 32bit int will take a max value of 2147483647, which only gives enough range for MMDDHHMMSS. I suspect the original author found that things weren't quite as they expected and just trimmed off the year.

If you want to create a patch for this, then you could convert frame to int64_t (may need to include stdint.h). However that will also mean the format string for the filename has to change, and that is provided by the user so will break every single app that uses the datetime option. 
Backwards compatibility must be maintained here as the app is widely used, therefore the only real fix is to update the help string - sorry.

mfn, 2015-12-09:
> Backwards compatibility must be maintained here as the app is widely used, therefore the only real fix is to update the help string - sorry.

No problem, I completely understand this!

I guess producing a working patch is out of my league (feature wise); however if someone would do that, maybe an approach would be to add a new option, like `--datetime-for-real` (just kidding, but the idea ...)

pelwell, 2015-12-09:
You could always rename it afterwards. It would be trivial to write a script that would do the job, and as long as you run it within a year of capture you'll get the right result.

mfn, 2015-12-09:
> You could always rename it afterwards. It would be trivial to write a script that would do the job, and as long as you run it within a year of capture you'll get the right result.

Absolutely; it's just a bit of hassle and having absolute non-ambiguous dates OOTB would be very nice.

6by9, 2016-01-01:
Change merged to correct the help text. Please close unless you think there is some further work required. 
If someone can come up with a PR to extend the field to actually include the year without breaking backward compatibility, then that would be great :-)

mfn, 2016-01-01:
I can only think about a new parameter to not break BC, so closing this.

Does it make sense to create a new issue for the "feature" or is it rather pointless without PR? :-)

closed 2016-01-01 by mfn

6by9, 2016-01-02:
Bring it up in the forums - you'll only tend to get a few people browsing the issues here on Github. 

Something similar has just come up as https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=130825 where they want to add milliseconds to the filename.