raspberrypi/userland #275: Userland build issue.

bhavinladhad, 2015-12-21:
I have built the userland repository for raspi.
The camera work fine. But when I run the mmal test examples. If get the following error  .vc.ril.camera  not found. 
Reading through the code I found the issue is with the constructor  supplier.  But did not find any code for the same.  Did I don't mistake while compiling  if so then y is raspistill work and not mmal_exapmles_basic 1 not work. or something else.

popcornmix, 2015-12-21:
So does your build of raspivid/raspivid work okay?
Not really sure what the mmal test examples do. @6by9 may know.

6by9, 2015-12-21:
Which test app are you trying to run? mmalplay/mmalcam? Sorry, not ones I've run recently and they're not released by default with Raspbian.
Is it just a typo in your error that there is a . before vc?

bhavinladhad, 2015-12-22:
The examples are basic operations of mmal on which raspi cam runs. They are located in the interface/mmal/test/examples, I don't think that they are released but they are there never the less in the source code.

bhavinladhad, 2015-12-22:
While checking through the code I found that bcm_host_init()had not been called in the examples. How can call that function in the examples and link the respective lib in the make or cmake files so that builder me can build the same. Kindly advice...

6by9, 2015-12-22:
OK, can take a look. 
As you can tell I haven't looked at those for a very long time - not sure why they suddenly got added in Sep2014, but c'est la vie.

Which app are you trying that uses the camera? mmalcam, except that that isn't in that directory? All the actual examples are doing video decode it seems.

bhavinladhad, 2015-12-22:
Trying all of the apps... Not one  working...  Get vc.ril.camera not found

bhavinladhad, 2015-12-22:
Apps ies example 1 2 and 3. Mmalcam mmalplay not working

6by9, 2015-12-22:
They weren't linking to mmal_vc_client, hence couldn't find any VC services. 
mmal_example_basic1 does something sending a load of buffers to/from video_decode. 
mmalplay works and renders stuff to the screen.
I haven't investigated the others. 
TBH I'm not sure how useful any of them are - might actually suggest purging them.

Commit pushed to https://github.com/6by9/userland/commit/55f6f35bb59268f0b446d2c596eaa80e5cc2bbd2 that fixes up the basics. I'll create a PR in a moment.

bhavinladhad, 2015-12-29:
I am trying to use the pi cam to sort ground nuts. Needed something fast.. Needed a direct mmal access buffers so that u can process the data instead of going through the processing of... Need to mmal video to process.... Kindly suggest any alternative if any...

6by9, 2015-12-29:
Raspividyuv - https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiVidYUV.c
It's the same as raspivid, but you'll get the raw frames coming in via camera_buffer_callback (line 665) so you're able to do as you please with the pixels from there.

6by9, 2015-12-31:
Changes merged to fix the test apps. Please close unless you feel there is more needed on those apps.

closed 2016-01-04 by bhavinladhad