raspberrypi/userland #281: Rendering errors with vertex coalescing and advanced stencilling.

Criffer, 2016-01-07:
glDrawArrays and glDrawElements appear to coalesce multiple vertices to avoid repeated drawing. (glxx_client.c:draw_arrays_or_elements). When stencilling is enabled with an advanced mode such as GL_INCR or GL_DECR, pixels can be written an incorrect number of times, causing the stencil buffer to not contain the intended result.

QPainter uses advanced stencilling to render non-convex paths. The path is rendered as a series of triangles from a common point; concave sections are written repeatedly, setting and clearing the stencil buffer repeatedly. The final state of concave sections should be clear. https://web.archive.org/web/20150328200622/http://blog.qt.io/blog/2010/01/06/qt-graphics-and-performance-opengl

I've attached a test-case generated from QPainter, which shows the bug. It comes from part of an SVG drawing. http://ariya.github.io/svg/tiger.svg

[testcase.zip](https://github.com/raspberrypi/userland/files/81458/testcase.zip)

popcornmix, 2016-01-20:
I'm afraid we are unlikely to be able to fix this. The original authors of the gpu 3d driver are gone. While we have the source and may be able to fix trivial issues, this sounds like something quite involved.

There is some good news. Eric Anholt is working on an open source arm-side [Gallium3D](http://www.phoronix.com/scan.php?page=news_item&px=MTcyMzE) driver. That has recently been merged into kernel: https://github.com/raspberrypi/linux/pull/1261

We are hoping to have a raspbian image that supports this in the next month or so. It is still very much alpha, but it may provide a solution for your bug.

trilkk, 2016-02-23:
I actually ran into this issue recently. Prevents rendering proper stencil shadows.

Have you found any way to avoid the issue? Even at the cost of performance.

christinabrien, 2016-02-26:
I have a workaround, and it will cost performance. The idea is to split draw calls for multiple primitives into individual draw calls. It prevents vertices from being coalesced in the driver, and does the correct number of stencil ops.

[glDrawWorkaround.c.txt](https://github.com/raspberrypi/userland/files/148115/glDrawWorkaround.c.txt)

Overriding GL_TRIANGLES and GL_TRIANGLE_FAN is sufficient for my purposes. You will probably want to extend it.

Metzelmaennchen, 2016-03-16:
Same problem when using two sided stencils for example in Doom3. The hint to the open source driver is nice, but how does this help on raspi1/zero when using a two sided stencil app there?

JamesH65, 2017-12-05:
Closing as a wont fix. The new driver should help here.

closed 2017-12-05 by JamesH65

christinabrien, 2017-12-05:
Is that the new driver that was ready "in the month or so" two years ago?

https://github.com/raspberrypi/userland/issues/281#issuecomment-173212660

pelwell, 2017-12-05:
Raspbian images have supported VC4 for over a year now, although it is not yet enabled by default.