raspberrypi/userland #284: requesting an intra frame while recording

chetanbnaik, 2016-01-22:
Hello Developers,

I am using picamera python module to capture video stream. I want to request a key frame (intra frame) while video recording is in progress. What is the correct way of doing this through mmal?

Many thanks for answering this quite a basic question.

JamesH65, 2016-01-22:
Please ask this sort of question on the Raspberry Pi forums - there is a camera section for exactly this sort of thing.

6by9, 2016-01-22:
Plus if you're using picamera then you should really be looking at https://github.com/waveform80/picamera
From MMAL, you're after MMAL_PARAMETER_VIDEO_REQUEST_I_FRAME - https://github.com/raspberrypi/userland/blob/master/interface/mmal/mmal_parameters_video.h#L54

closed 2016-01-23 by popcornmix