raspberrypi/userland #288: What is MMAL_COMPONENT_DEFAULT_MIRACAST?

Terminus-IMRC, 2016-02-08:
c.f. https://github.com/raspberrypi/userland/blob/master/interface/mmal/util/mmal_default_components.h#L68

What is ```MMAL_COMPONENT_DEFAULT_MIRACAST``` and how do I use it?
Or is it just an internal function?

6by9, 2016-02-09:
Not built on Pi.
For other projects there was a single component that would create a miracast encoded video stream, with various optimisations. I'm not certain all the relevant hooks are even present on Pi.
Ignore it.

Terminus-IMRC, 2016-02-09:
Thank you for your reply.
I hope I can use it also on Pi...!

Terminus-IMRC, 2016-02-10:
FYI: "miracast" strings were first seen in https://github.com/raspberrypi/firmware/commit/d3ada6f943818eecda9a81c1091631d3971798b2 .

closed 2016-02-09 by Terminus-IMRC