raspberrypi/userland #291: build fails with clang 3.8.0 on FreeBSD armv6

MikaelUrankar, 2016-02-15:
Hi,

We are seeing this error on FreeBSD armv6 with clang 3.8.0:
```
[ 21%] Building C object interface/khronos/CMakeFiles/EGL_static.dir/vg/vg_int_mat3x3.c.o
/usr/obj/portswork/usr/ports/misc/raspberrypi-userland/work/userland-fcbf8a3/interface/mmal/util/mmal_param_convert.c:47:7: error: variable 'i' is incremented both in the loop header and in the loop body [-Werror,-Wfor-loop-analysis]
      i++;
      ^
/usr/obj/portswork/usr/ports/misc/raspberrypi-userland/work/userland-fcbf8a3/interface/mmal/util/mmal_param_convert.c:40:25: note: incremented here
   for (i=0; i<n_pairs; i++)
                        ^
1 error generated.
```

The issue is described in details here:
https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=207097

6by9, 2016-02-15:
Yes, looks like a genuine bug. Feel free to create a pull request!

popcornmix, 2016-02-18:
This is now fixed.

closed 2016-02-18 by popcornmix