raspberrypi/userland #302: dtoverlay fails to compile (master branch)

fordfrog, 2016-03-30:
[100%] Building C object host_applications/linux/apps/dtoverlay/CMakeFiles/dtoverlay.dir/dtoverlay_main.c.o
cd /var/tmp/portage/media-libs/raspberrypi-userland-0_pre20160316/work/raspberrypi-userland-0_pre20160316_build/host_applications/linux/apps/dtoverlay && /usr/bin/armv7a-hardfloat-linux-gnueabi-gcc  -DEGL_SERVER_DISPMANX -DHAVE_CMAKE_CONFIG -DHAVE_VMCS_CONFIG -DOMX_SKIP64BIT -DTV_SUPPORTED_MODE_NO_DEPRECATED -DUSE_VCHIQ_ARM -DVCHI_BULK_ALIGN=1 -DVCHI_BULK_GRANULARITY=1 -D_FILE_OFFSET_BITS=64 -D_HAVE_SBRK -D_LARGEFILE64_SOURCE -D_LARGEFILE_SOURCE -D_REENTRANT -D__VIDEOCORE4__ -I/var/tmp/portage/media-libs/raspberrypi-userland-0_pre20160316/work/raspberrypi-userland-2f56a29/build/inc -I/var/tmp/portage/media-libs/raspberrypi-userland-0_pre20160316/work/raspberrypi-userland-2f56a29/host_applications/framework -I/var/tmp/portage/media-libs/raspberrypi-userland-0_pre20160316/work/raspberrypi-userland-2f56a29 -I/var/tmp/portage/media-libs/raspberrypi-userland-0_pre20160316/work/raspberrypi-userland-2f56a29/interface/vcos/pthreads -I/var/tmp/portage/media-libs/raspberrypi-userland-0_pre20160316/work/raspberrypi-userland-2f56a29/interface/vmcs_host/linux -I/var/tmp/portage/media-libs/raspberrypi-userland-0_pre20160316/work/raspberrypi-userland-2f56a29/interface/vmcs_host -I/var/tmp/portage/media-libs/raspberrypi-userland-0_pre20160316/work/raspberrypi-userland-2f56a29/interface/vmcs_host/khronos -I/var/tmp/portage/media-libs/raspberrypi-userland-0_pre20160316/work/raspberrypi-userland-2f56a29/interface/khronos/include -I/var/tmp/portage/media-libs/raspberrypi-userland-0_pre20160316/work/raspberrypi-userland-0_pre20160316_build -I/var/tmp/portage/media-libs/raspberrypi-userland-0_pre20160316/work/raspberrypi-userland-2f56a29/interface/vchiq_arm -I/var/tmp/portage/media-libs/raspberrypi-userland-0_pre20160316/work/raspberrypi-userland-2f56a29/host_support/include -I/var/tmp/portage/media-libs/raspberrypi-userland-0_pre20160316/work/raspberrypi-userland-2f56a29/interface/mmal -I/var/tmp/portage/media-libs/raspberrypi-userland-0_pre20160316/work/raspberrypi-userland-2f56a29/opensrc/helpers/libfdt -I/var/tmp/portage/media-libs/raspberrypi-userland-0_pre20160316/work/raspberrypi-userland-2f56a29/helpers/dtoverlay   -DNDEBUG -O2 -pipe -march=armv7-a -mfpu=vfpv3-d16 -mfloat-abi=hard -ggdb  -Wno-multichar -Wall -Wno-unused-but-set-variable -fPIC   -Wall -Werror -o CMakeFiles/dtoverlay.dir/dtoverlay_main.c.o -c /var/tmp/portage/media-libs/raspberrypi-userland-0_pre20160316/work/raspberrypi-userland-2f56a29/host_applications/linux/apps/dtoverlay/dtoverlay_main.c
/var/tmp/portage/media-libs/raspberrypi-userland-0_pre20160316/work/raspberrypi-userland-2f56a29/host_applications/linux/apps/dtoverlay/dtoverlay_main.c: In function ‘overlay_applied’:
/var/tmp/portage/media-libs/raspberrypi-userland-0_pre20160316/work/raspberrypi-userland-2f56a29/host_applications/linux/apps/dtoverlay/dtoverlay_main.c:636:2: error: ignoring return value of ‘fread’, declared with attribute warn_unused_result [-Werror=unused-result]
  fread(status, sizeof(status), 1, fp);
  ^
/var/tmp/portage/media-libs/raspberrypi-userland-0_pre20160316/work/raspberrypi-userland-2f56a29/host_applications/linux/apps/dtoverlay/dtoverlay_main.c: In function ‘dtoverlay_remove’:
/var/tmp/portage/media-libs/raspberrypi-userland-0_pre20160316/work/raspberrypi-userland-2f56a29/host_applications/linux/apps/dtoverlay/dtoverlay_main.c:356:5: error: ignoring return value of ‘chdir’, declared with attribute warn_unused_result [-Werror=unused-result]
     chdir(work_dir);
     ^
cc1: all warnings being treated as errors
host_applications/linux/apps/dtoverlay/CMakeFiles/dtoverlay.dir/build.make:62: recipe for target 'host_applications/linux/apps/dtoverlay/CMakeFiles/dtoverlay.dir/dtoverlay_main.c.o' failed
make[2]: *** [host_applications/linux/apps/dtoverlay/CMakeFiles/dtoverlay.dir/dtoverlay_main.c.o] Error 1
make[2]: Leaving directory '/var/tmp/portage/media-libs/raspberrypi-userland-0_pre20160316/work/raspberrypi-userland-0_pre20160316_build'
CMakeFiles/Makefile2:4554: recipe for target 'host_applications/linux/apps/dtoverlay/CMakeFiles/dtoverlay.dir/all' failed
make[1]: *** [host_applications/linux/apps/dtoverlay/CMakeFiles/dtoverlay.dir/all] Error 2
make[1]: Leaving directory '/var/tmp/portage/media-libs/raspberrypi-userland-0_pre20160316/work/raspberrypi-userland-0_pre20160316_build'
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

clivem, 2016-03-30:
Don't think this has made it from pull request into the repo yet....

https://github.com/raspberrypi/userland/pull/300/commits/60ba26004c85ca2d730da771ea5a2e89e8149330

pelwell, 2016-03-30:
A patch for this, https://github.com/raspberrypi/userland/issues/301 and https://github.com/raspberrypi/userland/pull/300 is in the pipeline.

pelwell, 2016-03-30:
Fixed by the latest commit.

pelwell, 2016-03-30:
Please test and close if happy.

clivem, 2016-03-30:
/home/fedora/rpmbuild/BUILD/raspberrypi-userland-964deab/host_applications/linux/apps/dtoverlay/dtoverlay_main.c: In function 'overlay_applied':
/home/fedora/rpmbuild/BUILD/raspberrypi-userland-964deab/host_applications/linux/apps/dtoverlay/dtoverlay_main.c:636:2: error: ignoring return value of 'fread', declared with attribute warn_unused_result [-Werror=unused-result]
  (void)fread(status, sizeof(status), 1, fp);

pelwell, 2016-03-30:
Which compiler are you using?

clivem, 2016-03-30:
[fedora@IQPlus3B1 SPECS]$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/armv7hl-redhat-linux-gnueabi/5.3.1/lto-wrapper
Target: armv7hl-redhat-linux-gnueabi
Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,objc,obj-c++,fortran,ada,go,lto --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-linker-hash-style=gnu --enable-plugin --enable-initfini-array --disable-libgcj --with-isl --disable-libmpx --enable-gnu-indirect-function --disable-sjlj-exceptions --with-tune=cortex-a8 --with-arch=armv7-a --with-float=hard --with-fpu=vfpv3-d16 --with-abi=aapcs-linux --build=armv7hl-redhat-linux-gnueabi
Thread model: posix
gcc version 5.3.1 20151207 (Red Hat 5.3.1-2) (GCC)

clivem, 2016-03-30:
The "int bytes __attribute__((unused)); bytes =....." was the patch I was using. gcc5 is not going to be fooled into ignoring the "unused result" by "(void)", apparently. ;)

pelwell, 2016-03-30:
I think it's a bit much for the compiler to demand the use of a GNU extension when a perfectly good, portable syntax existed. Anyway, the latest update actually uses the result.

clivem, 2016-03-30:
LOL. No compiler can complain about actually using the result....... (Famous last words.)

pelwell, 2016-03-30:
Insufficient interest in the result?

clivem, 2016-03-30:
I'll suggest, "-Winsufficient-interest", to the gcc devs and see what they think. ;)

fordfrog, 2016-03-30:
with the latest fix it compiles fine. thanks for the fix :-) i'm happy (and i hope clivem is too) so closing this bug.

closed 2016-03-30 by fordfrog

clivem, 2016-03-30:
I'm always happy! But now I'm being nosey.... Noticed "/var/tmp/portage/...." Portage.... That's Gentoo, right? Does that mean there is a Gentoo distro for Pi, or just that you are cross-compiling on a Gentoo desktop?

fordfrog, 2016-03-30:
yes, that's gentoo. gentoo is only one, there are no different distros for different archs. gentoo supports various archs including arm. i'm compiling the stuff directly on rpi2 (no cross-compiling, distcc etc). i compiled there whole system.

i am gentoo dev. i'm trying to make kodi to work on gentoo on rpi. in fact it already works with omxplayer. we have another issue with mmal that i reported just few minutes back.

clivem, 2016-03-30:
Ah, OK. I think you should treat yourself to a Pi3 if you are compiling everything on Pi. It'll reduce the wait time. Just to give you an idea of the potential benefit, ISTR, kernel compile time is something like 2.5 hours on a 2B and only 1.5 hours on 3B.

fordfrog, 2016-03-30:
i was thinking about that too but i have 8 rpi2 boxes here that i need the same installation on so i need to build stuff on rpi2. but maybe i will buy rpi3 too to speed up some testing etc so i install on rpi2 just the already tested stuff which would definitely save me some time :-)

clivem, 2016-03-30:
Well, you'll be using exactly the same 32 bit image on the 3B, you'll mostly benefit from the increased core speed of the A53 cores. There is no arch change, or 64 bit.... No compatibilty issues. Anything you compile on 3B using your current 2B image will work just fine when run back on 2B.

fordfrog, 2016-03-30:
yeah, that could work. the only differences i found is the chip (BCM2836 vs BCM2837), processor (Cortex-A7 vs Cortex-A53) and that extra wireless/bluetooth so it could really work. good idea, thanks! :-)