raspberrypi/userland #303: linking issue - mmal: mmal_component_create_core: could not find component 'vc.ril.video_render' (master branch)

fordfrog, 2016-03-30:
this error occurs when trying to play video in kodi using dvdplayer.

here are mmal libs that kodi links against:
# ldd -d /usr/lib/kodi/kodi.bin | grep mmal
        libmmal.so => /usr/lib/opengl/raspberrypi-userland/lib/libmmal.so (0x76b5a000)
        libmmal_core.so => /usr/lib/opengl/raspberrypi-userland/lib/libmmal_core.so (0x76b39000)
        libmmal_util.so => /usr/lib/opengl/raspberrypi-userland/lib/libmmal_util.so (0x76b19000)

i'm not that familiar with linking stuff, but there are some libraries with undefined symbols. executables matching /opt/vc/bin/ and /opt/vc/lib/plugins/*so are ok but there are issues for /opt/vc/lib/*so (including complete list):

# ldd -d /opt/vc/lib/*so
/opt/vc/lib/libbcm_host.so:
        linux-vdso.so.1 (0x7ef41000)
        libvchiq_arm.so => /usr/lib/opengl/raspberrypi-userland/lib/libvchiq_arm.so (0x76f76000)
        libvcos.so => /usr/lib/opengl/raspberrypi-userland/lib/libvcos.so (0x76f5a000)
        libpthread.so.0 => /lib/libpthread.so.0 (0x76f32000)
        libc.so.6 => /lib/libc.so.6 (0x76dfc000)
        /lib/ld-linux-armhf.so.3 (0x54b7a000)
        libdl.so.2 => /lib/libdl.so.2 (0x76de9000)
        librt.so.1 => /lib/librt.so.1 (0x76dd2000)
/opt/vc/lib/libcontainers.so:
        linux-vdso.so.1 (0x7efac000)
        libvcos.so => /usr/lib/opengl/raspberrypi-userland/lib/libvcos.so (0x76f42000)
        libpthread.so.0 => /lib/libpthread.so.0 (0x76f1a000)
        libc.so.6 => /lib/libc.so.6 (0x76de5000)
        /lib/ld-linux-armhf.so.3 (0x54b00000)
        libdl.so.2 => /lib/libdl.so.2 (0x76dd1000)
        librt.so.1 => /lib/librt.so.1 (0x76dba000)
/opt/vc/lib/libdebug_sym.so:
        linux-vdso.so.1 (0x7ef9f000)
        libc.so.6 => /lib/libc.so.6 (0x76dc0000)
        /lib/ld-linux-armhf.so.3 (0x54b33000)
undefined symbol: vcos_log_impl (/opt/vc/lib/libdebug_sym.so)
undefined symbol: vcos_log_register     (/opt/vc/lib/libdebug_sym.so)
/opt/vc/lib/libdtovl.so:
        linux-vdso.so.1 (0x7efcc000)
        libc.so.6 => /lib/libc.so.6 (0x76dfd000)
        /lib/ld-linux-armhf.so.3 (0x54b85000)
undefined symbol: fdt_first_property_offset     (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_get_path  (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_check_header      (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_get_name  (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_appendprop        (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_setprop_inplace   (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_next_subnode      (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_get_phandle       (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_get_property      (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_setprop   (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_add_subnode_namelen       (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_subnode_offset_namelen    (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_getprop   (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_delprop   (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_path_offset       (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_subnode_offset    (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_pack      (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_node_offset_by_phandle    (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_path_offset_namelen       (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_next_node (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_first_subnode     (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_next_property_offset      (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_getprop_by_offset (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_getprop_namelen   (/opt/vc/lib/libdtovl.so)
undefined symbol: fdt_del_node  (/opt/vc/lib/libdtovl.so)
/opt/vc/lib/libEGL.so:
        linux-vdso.so.1 (0x7ef10000)
        libbcm_host.so => /usr/lib/opengl/raspberrypi-userland/lib/libbcm_host.so (0x76ebe000)
        libm.so.6 => /lib/libm.so.6 (0x76e45000)
        libvchiq_arm.so => /usr/lib/opengl/raspberrypi-userland/lib/libvchiq_arm.so (0x76e2e000)
        libvcos.so => /usr/lib/opengl/raspberrypi-userland/lib/libvcos.so (0x76e11000)
        libpthread.so.0 => /lib/libpthread.so.0 (0x76de9000)
[... 234 lines cut ...]
/opt/vc/lib/libvcsm.so:
        linux-vdso.so.1 (0x7eed2000)
        libvcos.so => /usr/lib/opengl/raspberrypi-userland/lib/libvcos.so (0x76f0d000)
        libpthread.so.0 => /lib/libpthread.so.0 (0x76ee5000)
        libc.so.6 => /lib/libc.so.6 (0x76db0000)
        /lib/ld-linux-armhf.so.3 (0x54b0d000)
        libdl.so.2 => /lib/libdl.so.2 (0x76d9c000)
        librt.so.1 => /lib/librt.so.1 (0x76d85000)
/opt/vc/lib/libvchiq_arm.so:
        linux-vdso.so.1 (0x7ebfa000)
        libvcos.so => /usr/lib/opengl/raspberrypi-userland/lib/libvcos.so (0x76ee6000)
        libpthread.so.0 => /lib/libpthread.so.0 (0x76ebe000)
        libc.so.6 => /lib/libc.so.6 (0x76d89000)
        /lib/ld-linux-armhf.so.3 (0x54b65000)
        libdl.so.2 => /lib/libdl.so.2 (0x76d75000)
        librt.so.1 => /lib/librt.so.1 (0x76d5e000)
/opt/vc/lib/libWFC.so:
        linux-vdso.so.1 (0x7ef5c000)
        libvchiq_arm.so => /usr/lib/opengl/raspberrypi-userland/lib/libvchiq_arm.so (0x76fb1000)
        libvcos.so => /usr/lib/opengl/raspberrypi-userland/lib/libvcos.so (0x76f95000)
        libpthread.so.0 => /lib/libpthread.so.0 (0x76f6d000)
        libc.so.6 => /lib/libc.so.6 (0x76e37000)
        /lib/ld-linux-armhf.so.3 (0x54aee000)
        libdl.so.2 => /lib/libdl.so.2 (0x76e24000)
        librt.so.1 => /lib/librt.so.1 (0x76e0d000)

6by9, 2016-03-30:
Taking just your first couple of lines, that sounds like a duplicate of https://github.com/raspberrypi/userland/issues/178 - the optimiser has been too thorough and removed the constructor that registers the VideoCore MMAL components.

I'm not sure what you're trying to impart with the linker dump.

fordfrog, 2016-03-30:
well, i suppose that "undefined symbol" is probably not ok, at least in some cases. but i'm not familiar with how linker works so maybe it is correct. anyway, i was able to remove some of the issues by modifying (or adding) target_link_libraries, like "target_link_libraries(dtovl fdt)", "target_link_libraries (mmal_core mmal_util vcos)" etc. i also found out that mmal_core needs mmal_util and mmal_util needs mmal_core which cmake did not like when i added the missing libraries. so as i do not understand this stuff, i'd rather have someone else to have a look at it and provide appropriate fix.

pelwell, 2016-03-30:
@fordfrog You are right. At least in the case of libdtovl, the libfdt.a archive should be built into the shared library, rather than being an unresolved dependency.

fordfrog, 2016-03-30:
@pelwell any idea, what's causing that mmal error? is it related to the undefined symbols or the issue is different?

pelwell, 2016-03-30:
I believe all of the mmal undefined symbols are expected, in that they are exported by other shared libraries.

@6by9's explanation sounds plausible. MMAL uses the "constructor" attribute to mark functions that should be called before entering main(). If that mechanism is failing then the components will not be registered, which will lead to the error you saw. Note that this is a runtime error, not a linking error.

6by9, 2016-03-30:
What exactly have you done? Steps to reproduce please.
Is this the Kodi installed from apt-get, or one you've compiled yourself?

It appears that Kodi isn't asking for libmmal_vc_client.so, and that is what will register the VideoCore MMAL components via the constructor (see #178). That's probably an issue with your build of Kodi, not the userland libraries themselves.

fordfrog, 2016-03-30:
* kodi 16.0, built on gentoo linux distro using kodi ebuild (bulding script)
* starting kodi, omxplayer disabled (in video acceleration menu section), playing video file (no picture, log file containing mmal errors)

* kodi really links only against three mmal libraries (mentioned in first post), mmal_client_vc not being one of them, if that is what you mean.
* mmal related stuff is handled in kodi build system, i found no reference to mmal in the ebuild itself so there is no extra configuration performed

do i get it right that kodi should use mmal_vc_client instead of mmal, mmal_core and mmal_util?

6by9, 2016-03-30:
> do i get it right that kodi should use mmal_vc_client instead of mmal, mmal_core and mmal_util?

As well as.
mmal_core provides the core functionality.
mmal_util is helper functions
mmal_vc_client adds support for talking to VideoCore for hardware acceleration.

Please read #178, it even links to https://github.com/plexinc/OpenELEC.tv/commit/3e46d682e8bda34ff663ade0d2cd21f6f8893b39 which is the OpenELEC change to amend the linker flags. You probably want something similar.

ED6E0F17, 2016-03-30:
I think this is the same thing on Gentoo:

https://github.com/srcshelton/gentoo-ebuilds/blob/master/media-libs/raspberrypi-userland/raspberrypi-userland-9999.ebuild#L23

fordfrog, 2016-03-30:
@ED6E0F17 yes, it is. i was told elsewhere that removing --as-needed is not a fix but workaround, so i was looking for another solution for kodi but found none. compiling c stuff is not what i am experienced with.

6by9, 2016-04-01:
Summary: This is a duplicate of #178, and should be closed as such.

If the community wishes to come up with an alternative then please put it forward.

The requirements are that:
- libmmal can be built independently of the component providers.
- somewhere, something has to load the component provider and trigger them to register the components that they support. This is currently done via the constructor attribute. mmal_vc_client is NOT a dependency on libmmal or libmmal_core - MMAL has been built for standalone Windows where VideoCore is nowhere in sight.

The only thing that really knows it wants to use VideoCore accelerated components is your application, so that is really what has to signal that it wants libmmal_vc_client.
Propose your solution without breaking any existing applications.

pelwell, 2016-04-01:
On the VPU the way I got round this was to have some key function like a constructor put in a ".init.*" section - see interface/vchi/vchi_autoservice.h for an example; this works because the linker never discards things in .init and init.*. I don't know if that technique works with the GNU linker, but it might be worth a try.

popcornmix, 2016-09-06:
Closing as duplicate of #178

closed 2016-09-06 by popcornmix