raspberrypi/userland #305: Slow handling of rapid MMAL_EVENT_PARAMETER_CHANGED events means you stop getting such events

malc0, 2016-04-06:
If camera_control_callback() (in RaspiVid.c) is slow -- for instance by redirecting stderr to a slow file and hitting the buffer flush during camera_control_callback()) -- and there are frequent calls to this function (by running with `-set` and `-nopreview` to destabilize the AWB loop) the camera component's event_pool runs out of buffers (by default mmal_component.c sets this to 4).

This results in:
```
mmal: mmal_port_event_get: vc.ril.camera(1:0) port 0x132b8a0, no event buffer left for EPCH
mmalipc: mmal_vc_handle_event_msg: no event buffer available to receive event (2)
```

All of this is fine and understandable.  The problem is that once you've had four `no event buffer left for EPCH' errors you never get any more EPCH events handed over from the host, even if there's plenty of client-side buffers available.  This is a problem for long-running programmes wanting to log the gains applied by the camera pipeline.

Further investigation shows the MMAL_EVENT_PARAMETER_CHANGED_T data is 740 bytes and hence transferred by vchiq_queue_bulk_receive(), rather than a memcpy() from the MMAL event space.  What follows is speculation, but I wonder if this is related to mmal_vc_handle_event_msg()'s `FIXME: How to abort bulk receive if necessary?' -- is the host expecting to do a transmit, but as there's no buffer the transmit never happens, and after four slots are consumed the notification mechanism on the host gets bunged up?

malc0, 2016-04-26:
Workaround: change MMAL_CONTROL_PORT_BUFFERS_MIN from 4 to 40, in interface/mmal/core/mmal_component.c, and recompile libmmal_core.so.

Obviously this is a complete hack, but posted lest anyone else wants to log gain settings over more than a few hours.

JamesH65, 2017-12-05:
We have a workaround, and I suspect engineering time is not available to spend time looking at any underlying cause, so flagging as a wont fix. Please reopen if necessary.

closed 2017-12-05 by JamesH65