raspberrypi/userland #307: bcm_host_deinit() required

mihalop, 2016-04-14:
I am using multiple processes interfacing with DispmanX API to create and use OpenGLES surfaces. Right now, we are using pi3d. These processes continue to run concurrently, and I need more than 4. But after bcm_host_init() is called internally by pi3d, there is no way to patch pi3d to disconnect from the service.

I have seen #197. In our case, calling bcm_host_deinit() after pi3d gets the surface is required. Any chance to provide an implementation based on the process exit clean up code?

popcornmix, 2016-04-14:
You say multiple processes. How many processes do you need?
(I'm assuming you are talking about processes, i.e. showing up with different pid's in top, rather than threads).
What GPU features do the processes use? (e.g. dispmanx, GLES2, gencmd, tvservice, cec, openmax, mmal etc)
What is the process doing after the point you want to call bcm_host_deinit()? Will it never use the GPU again until it exits?

mihalop, 2016-04-14:
Yes, they are different processes, not threads. A limit would be something like 10 or 15. They are different python apps, based on pi3d, which under the hood uses GLES2, and dispmanx to obtain the gl surface. Each app draws each on each own surface.

These processes continuously use opengles calls. I though that perhaps I could get the gl surface, disconnect the services connected through the bcm_init, and then continue with only opengles calls, till the process exited. Perhaps I got something wrong.

Currently, I hit the limit of 4 processes. And I can't exactly figure out how to bypass it with different processes.

popcornmix, 2016-04-15:
You don't need to call bcm_host_init. Make a copy of https://github.com/raspberrypi/userland/blob/master/host_applications/linux/libs/bcm_host/bcm_host.c#L83 and remove initialisation of services you don't need (probably gencmd, tv and cec).
Call your version of bcm_host_init to start and call vc_dispmanx_stop when you have finished with dispmanx. That may achieve what you need.

mihalop, 2016-04-18:
Made several tests. Stopping dispmanx kills all opengles surfaces that have been created. Tried it both on a single process as well as across processes (a master process makes the dispmanx calls while other processes draw on reurned opengles surfaces). So, this probably does not work.

It just seems that you just can't have more than 3-4 processes doing opengles calls. Will be implementing the app differently, providing a simple wrapper and calling all different apps (as modules) from a single drawing loop.

I wonder how Wayland works. Thanks for the feedback anyway.

popcornmix, 2016-04-18:
Did you check how many dispmanx clients can be registered? Looks like the limit is 8 (some of the other services like gencmd support fewer).

mihalop, 2016-04-19:
No. During tests I just called bcm_host_init and then shutdown dispmanx. Since it destroyed all created opengles surfaces, there was no point in digging in further. Even if it topped at 8 dispmanx connections, having only 8 processes is still not enough for what I am after.

popcornmix, 2016-04-19:
In that case you are probably doing something like Wayland.
I believe they have a "window manager" process which does the dispman calls and provides surfaces to other processes which just use GL.

JamesH65, 2017-12-05:
Any thing to add to this issue? Or is it closable?

mihalop, 2017-12-05:
We did implement a dispmanx "window manager" sort of thingy and got around this restriction. closing issue.

closed 2017-12-05 by mihalop