raspberrypi/userland #311: raspicam update for new camera

egoist-sx, 2016-04-29:
I have just came across that a updated version of raspberry pi camera is released. New camera uses Sony IMX219PQ as CMOS sensor which supports ISP HDR. After checking RaspiStill.c, it seems default to old OV5647 CMOS, and no exposed interface for HDR mode. 

Is HDR mode working in progress or in plan?

lurch, 2016-05-01:
https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=145815 says "HDR is NOT supported. The ISP doesn't have suitable processing to implement it."

And I believe https://github.com/raspberrypi/userland/commit/f824ef5530012f83a9ef8a8089eaadf13f6963e9 is where support was added for the new camera.

6by9, 2016-05-02:
HDR is NOT supported on Pi and is extremely unlikely ever to be added to the GPU driver as there is no hardware to implement it so performance suffers.
The sensor supporting writing out the alternating exposure time frames/lines is about 10% of the problem, processing those images into something useful is the other 90%

@popcornmix Please close as won't fix.

closed 2016-05-02 by popcornmix