raspberrypi/userland #313: hello_video: Resquest:  Define layer

fredericofs, 2016-05-05:
Please add a --layer option to the hello_video example.

popcornmix, 2016-05-05:
These are designed as simple examples of how to use the openmax api.
hello_video itself isn't designed as a video player, so adding command line options isn't what this is designed for.
There is a variant of hello_video here:
https://github.com/popcornmix/userland/tree/master/host_applications/linux/apps/hello_pi/hello_video_simple
which does configure the layer. Change this number here:
https://github.com/popcornmix/userland/blob/master/host_applications/linux/apps/hello_pi/hello_video_simple/video.c#L84
and rebuild.

fredericofs, 2016-05-05:
Okay I understand.
I see the layer number is a bit shift opration, I dont understand #C yet. Can I change this to a simple number?
Thank you!

closed 2016-05-05 by fredericofs

popcornmix, 2016-05-05:
Yes, layer is a signed 16-bit number with higher numbers being "on top".
(1<<15)-1 is the same as 32767 which is the largest number possible.