raspberrypi/userland #317: raspistill: Strange blue artifacts in dark portions of an image

shrx, 2016-06-10:
_Disclaimer: I have also posted this issue on the [RPi forums](https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=150054), however it received no attention._

I have noticed that sometimes, if the light is just weak enough, the camera will produce a strange blue tint over darker parts of the scene.
Please check the images in album to see what I mean: http://imgur.com/a/kTvyE
The first image, taken at 7 pm, shows no visible artifacts.
The second image, taken at 8 pm, shows some blue tint on the left and right edges in the dark areas.
The third image, taken at 9 pm, shows very pronounced artifacts. The whole bottom half of the image is too blue.
The last image, taken at 10 pm, again shows no visible artifacts.
The first three pics were taken with the command `raspistill -vf -hf -o pic-${DATETIME}.jpg`
The last pic was taken with the command `raspistill -vf -hf -ss 6000000 -ISO 800 -o pic-${DATETIME}.jpg`

System & hardware info:
Raspberry Pi B
Raspberry Pi Camera v1 with a wide-angle lens on top of the stock lens
Linux pi2 4.4.11+ #888 Mon May 23 20:02:58 BST 2016 armv6l GNU/Linux
Custom lines in /boot/config.txt:
start_x=1
gpu_mem=128
disable_camera_led=1
ov5647_acls=1

I suspect the `ov5647_acls=1` setting causes this anomaly because I have been using this setup for years before enabling it and I never had any problems. This artifacting also does not happen every day, so I suspect the lighting conditions have to be just right to trigger it.

6by9, 2016-06-10:
So disable ```ov5647_acls=1``` again. It is not enabled by default as it hasn't had the testing required to make that change.
You're also not using a standard Pi camera, so lens shading artifacts are expected.

The other recommendation is to add ```-st``` for a stats pass on the image to allow AE, AWB, and lens shading to all do the best job they can.

shrx, 2016-06-10:
I am using a standard Pi camera, it has not been modified in any way. It just looks through another, separate lens. I need the `ov5647_acls=1` parameter to help with the cyan vignetting that is otherwise present (even without the secondary lens).

6by9, 2016-06-10:
Lens shading is a factor of the total optics the camera is seeing through. By adding a second lens you've changed those optics and observe a different lens shading situation to the standard camera with unmodified optics.

```ov5647_acls=1``` modifies the handover of data from preview to stills. ```-st``` forces recomputation on the captured frame in the stills processing. They give the same end result over cyan vignetting, with --st being closer to ideal as it is working on the final frame.

Either way around they're working with a captured frame from an uncontrolled scene, so if any assumptions the algorithm makes over the colour composition of the frame are inaccurate, then it can produce incorrect results as any automatic processing can.
The only other way around this is to capture the raw Bayer images and you set up a postprocessing pipeline with you having produced calibrated lens shading compensation values for your lens setup.

shrx, 2016-06-10:
I tried with `ov5647_acls=1` disabled and `-st` enabled. Here is the comparison:

`ov5647_acls=1` enabled; `-st` disabled
[image]
`ov5647_acls=1` disabled; `-st` enabled
[image]

The second does indeed look better, however the greens might be a bit too red.

6by9, 2017-10-10:
Closing due to inactivity, and that you can now calibrate your own lens shading table for when using non-standard lenses - https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=190586

closed 2017-10-10 by 6by9