raspberrypi/userland #320: raspistill --framestart off by one error

mutability, 2016-06-18:
raspistill help claims:

-fs, --framestart	: Starting frame number in output pattern(%d)

However, if you provide (for example) --framestart 1234 then the first frame captured is frame 1235.

```
$ raspistill --nopreview --width 820 --height 616 --timelapse 2000 --timeout 10000 --output test_%04d.png --framestart 20 --encoding png --rotation 180 --annotate 12 -v
[...]
Connecting camera preview port to video render.
Connecting camera stills port to encoder input port
Opening output file test_0021.png
Enabling encoder output port
Starting capture 21
Finished capture 21
```

(libraspberrypi-bin 1.20160523-1)

waveform80, 2016-08-15:
After a quick skim of the source, I suspect this is due to [`wait_for_next_frame`](https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiStill.c#L1558) (or where it's called from). This function (obviously) waits for the next frame to be available according to whatever criteria have been specified (timeout, key-press, etc.) and also handles incrementing the frame number. It's called at the *top* of the [main loop](https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiStill.c#L1913) hence when `state->frameStart` is set to 20 (or 0 by default), it gets incremented prior to the first capture.

The call can't be moved to the bottom of the main loop or the frame won't necessarily be "ready" for the first capture. The frame-number incrementing responsibility could be removed from `wait_for_next_frame` and bunged in the main loop instead, but I suspect the simplest fix is just to decrement `frameStart` [after it's set](https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiStill.c#L571). I'm not sure if that would affect the other modes (timestamp as frame number) as so on (I don't *think* it does as the main loop appears to overwrite the frame number in those cases).

popcornmix, 2016-08-15:
@6by9 any preference for the fix?

6by9, 2016-08-15:
I'd vote for the minimal change.
Probably cleanest to do it at https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiStill.c#L1911

Do we care about wrapping? They're 32 bit signed values, so the behaviour should be consistent anyway.

We ought to fix https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiStill.c#L573 whilst at it though:
``` if (sscanf(argv[i + 1], "%u", &state->frameStart) == 1)```
when state->frameStart is an int, not unsigned.

popcornmix, 2016-08-15:
@6by9 is that what you meant? http://paste.ubuntu.com/23058734/

6by9, 2016-08-15:
Yup - exactly that. Sorry if I was unclear.