raspberrypi/userland #323: Requesting buffer size messes up client state

ldmnyblzs, 2016-07-13:
When `glGetBufferParameteriv` is called with `GL_BUFFER_SIZE`, it creates and stores a `GLXX_BUFFER_INFO_T` but sets `id` and `cached_size` only. `mapped_pointer` and `mapped_size` get random values. This makes every subsequent call to `glBufferData` fail and `glGetBufferPointervOES` also returns false information from this point on.

popcornmix, 2016-07-13:
We don't have anyone who really knows this code, and GL is being replaced with the open-source arm side driver. I'm just guessing, but this code does look a little suspicious:
https://github.com/raspberrypi/userland/blob/master/interface/khronos/glxx/glxx_client.c#L2110

Could you try inserting:
```glxx_buffer_info_get(state, target, &buffer);```
before the
``buffer.cached_size = params[0];``
line, and rebuild userland?

JamesH65, 2017-12-05:
@popcornmix no response from OP, but should we be thinking of testing this fix? Or just wont fix the issue?

JamesH65, 2018-07-02:
@timg236 Any thoughts?

timg236, 2018-07-02:
I think the current implementation is wrong. It's attempting to updated it's cached copy of the buffer size but should not be modifying mapped_size and mapped_values. 
As @popcornmix mentions the fix is probably to call info_get first and then modify just the cached size before calling info_set.

It's the legacy driver but the fix is simple and low risk.

JamesH65, 2018-07-02:
@timg236 Can you bung in a PR so it can be reviewed/merged?

PR #472 (2018-07-04): glxx_client: Fix glGetBufferParameteriv GL_BUFFER_SIZE
glGetBufferParameteriv updates the client side cached copy of the
buffer_size. However, when it updated the cache it by calling
glxx_buffer_info_set it trashed the mapped_pointer and mapped_size
values because they are in the same structure as the cached_size but
this was being allocated on the stack.

The trampled mapped_pointer / mapped_size could cause crashes or invalid
buffer data to be uploaded.

Change to read the current cached value and then modify just the
cached_size preserving whatever was in the cache before.

Fixes #323
commit 409dfcd9 (2018-07-04): glxx_client: Fix glGetBufferParameteriv GL_BUFFER_SIZE

glGetBufferParameteriv updates the client side cached copy of the
buffer_size. However, when it updated the cache it by calling
glxx_buffer_info_set it trashed the mapped_pointer and mapped_size
values because they are in the same structure as the cached_size but
this was being allocated on the stack.

The trampled mapped_pointer / mapped_size could cause crashes or invalid
buffer data to be uploaded.

Change to read the current cached value and then modify just the
cached_size preserving whatever was in the cache before.

Fixes #323
closed 2018-07-04 by popcornmix