raspberrypi/userland #326: HelloVideo not working for raspberry pi 3 on android platform

rockster4india, 2016-08-01:
Hi,

I have cross compiled user land application "hello video" for raspberry pi 3 on android(n preview 2).
There is no error while running the application but video(test.h264) is not playing 
PFA log below

rpi3:/storage/emulated/0 # hellovideo test.h264                                
Entering bcm_host_init
bcm_host_init success and call to video decode test
Video Decode Test log_trace
status from component : video_decode 0
status from component : video_render 0
status from component : clock0
status from component : video_scheduler 0
components are created 
wait for event in video render
Value from ilclient_wait_for_event -1
End of playback

rpi3:/storage/emulated/0 # logcat         
                                                                                                                                               }
01-01 00:07:16.084  1615  1615 I chatty  : uid=0(root) hellovideo expire 76 lines
01-01 00:07:16.087  1615  1619 I chatty  : uid=0(root) hellovideo expire 1 line
01-01 00:07:16.113  1615  1616 I chatty  : uid=0(root) hellovideo expire 79 lines
01-01 00:07:16.113  1615  1621 I chatty  : uid=0(root) hellovideo expire 183 lines
01-01 00:07:16.132  1615  1616 I chatty  : uid=0(root) hellovideo expire 85 lines
01-01 00:07:16.132  1615  1615 I chatty  : uid=0(root) hellovideo expire 174 lines
01-01 00:07:16.198  1615  1616 I chatty  : uid=0(root) hellovideo expire 85 lines
01-01 00:07:16.198  1615  1621 I chatty  : uid=0(root) hellovideo expire 82 lines
01-01 00:07:16.343  1615  1621 D vchiq_lib: vchiq_release_message handle=00000107, header=b6a8a810
01-01 00:07:16.343  1615  1621 I ilclient: cl:video_decode:0: empty buffer done 0xb6c212b0
01-01 00:07:16.348  1615  1615 D vchiq_lib: vchiq_queue_message called service handle = 0x00000107
01-01 00:07:16.348  1615  1615 D vchiq_lib: vchiq_queue_bulk_transmit_handle called service handle = 0x00000107
01-01 00:07:16.348  1615  1615 D vchiq_lib: vchiq_queue_bulk_transmit called service handle = 0x00000107
01-01 00:07:16.349  1615  1616 D vchiq_lib: callback(3, 0, b6ebbf8c(107,b6c32e10), 0)
01-01 00:07:16.349  1615  1616 D vchiq_lib: callback(2, b6a8d010, b6ebbf8c(107,b6c32e10), 0)
01-01 00:07:16.349  1615  1621 D vchiq_lib: vchiq_release_message handle=00000107, header=b6a8d010
01-01 00:07:16.362  1615  1616 D vchiq_lib: callback(2, b6a8a810, b6ebbf8c(107,b6c32e10), 0)
01-01 00:07:16.362  1615  1621 D vchiq_lib: vchiq_release_message handle=00000107, header=b6a8a810
01-01 00:07:16.362  1615  1621 I ilclient: cl:video_decode:0: empty buffer done 0xb6c21320
01-01 00:07:16.362  1615  1615 D vchiq_lib: vchiq_queue_message called service handle = 0x00000107
01-01 00:07:16.362  1615  1615 D vchiq_lib: vchiq_queue_bulk_transmit_handle called service handle = 0x00000107
01-01 00:07:16.362  1615  1615 D vchiq_lib: vchiq_queue_bulk_transmit called service handle = 0x00000107
01-01 00:07:16.363  1615  1616 D vchiq_lib: callback(3, 0, b6ebbf8c(107,b6c32e10), 0)
01-01 00:07:16.363  1615  1621 D vchiq_lib: vchiq_release_message handle=00000107, header=b6a8d010
01-01 00:07:16.380  1615  1616 D vchiq_lib: callback(2, b6a8a810, b6ebbf8c(107,b6c32e10), 0)
01-01 00:07:16.380  1615  1621 D vchiq_lib: vchiq_release_message handle=00000107, header=b6a8a810
01-01 00:07:16.380  1615  1621 I ilclient: cl:video_decode:0: empty buffer done 0xb6c21390
01-01 00:07:16.382  1615  1615 D vchiq_lib: vchiq_queue_message called service handle = 0x00000107
01-01 00:07:16.382  1615  1615 D vchiq_lib: vchiq_queue_bulk_transmit_handle called service handle = 0x00000107
01-01 00:07:16.382  1615  1615 D vchiq_lib: vchiq_queue_bulk_transmit called service handle = 0x00000107
01-01 00:07:16.382  1615  1616 D vchiq_lib: callback(3, 0, b6ebbf8c(107,b6c32e10), 0)
01-01 00:07:16.382  1615  1616 D vchiq_lib: callback(2, b6a8d010, b6ebbf8c(107,b6c32e10), 0)
01-01 00:07:16.382  1615  1621 D vchiq_lib: vchiq_release_message handle=00000107, header=b6a8d010
01-01 00:07:16.416  1615  1616 D vchiq_lib: callback(2, b6a8a810, b6ebbf8c(107,b6c32e10), 0)
01-01 00:07:16.416  1615  1621 D vchiq_lib: vchiq_release_message handle=00000107, header=b6a8a810
01-01 00:07:16.416  1615  1621 I ilclient: cl:video_decode:0: empty buffer done 0xb6c21400
01-01 00:07:16.417  1615  1615 D vchiq_lib: vchiq_queue_message called service handle = 0x00000107
01-01 00:07:16.417  1615  1615 D vchiq_lib: vchiq_queue_bulk_transmit_handle called service handle = 0x00000107
01-01 00:07:16.417  1615  1615 D vchiq_lib: vchiq_queue_bulk_transmit called service handle = 0x00000107
01-01 00:07:16.417  1615  1616 D vchiq_lib: callback(3, 0, b6ebbf8c(107,b6c32e10), 0)
01-01 00:07:16.417  1615  1616 D vchiq_lib: callback(2, b6a8d010, b6ebbf8c(107,b6c32e10), 0)
01-01 00:07:16.417  1615  1621 D vchiq_lib: vchiq_release_message handle=00000107, header=b6a8d010
01-01 00:07:16.417  1615  1616 D vchiq_lib: callback(2, b6a8a810, b6ebbf8c(107,b6c32e10), 0)
01-01 00:07:16.418  1615  1621 D vchiq_lib: vchiq_release_message handle=00000107, header=b6a8a810
01-01 00:07:16.418  1615  1621 I ilclient: cl:video_decode:0: empty buffer done 0xb6c21470
01-01 00:07:16.423  1615  1615 D vchiq_lib: vchiq_queue_message called service handle = 0x00000107
01-01 00:07:16.423  1615  1615 D vchiq_lib: vchiq_queue_bulk_transmit_handle called service handle = 0x00000107
01-01 00:07:16.423  1615  1615 D vchiq_lib: vchiq_queue_bulk_transmit called service handle = 0x00000107
01-01 00:07:16.423  1615  1616 D vchiq_lib: callback(3, 0, b6ebbf8c(107,b6c32e10), 0)
[... 225 lines cut ...]
01-01 00:07:32.445  1615  1621 D vchiq_lib: vchiq_release_message handle=00000107, header=b6a8bc10
01-01 00:07:32.445  1615  1621 I ilclient: cl:video_decode:0: callback state changed (Loaded)
01-01 00:07:32.445  1615  1621 D vchiq_lib: vchiq_release_message handle=00000107, header=b6a8a810
01-01 00:07:32.445  1615  1615 D vchiq_lib: vchiq_queue_message called service handle = 0x00000107
01-01 00:07:32.445  1615  1616 D vchiq_lib: callback(2, b6a89410, b6ebbf8c(107,b6c32e10), 0)
01-01 00:07:32.445  1615  1616 D vchiq_lib: callback(2, b6a8d010, b6ebbf8c(107,b6c32e10), 0)
01-01 00:07:32.445  1615  1621 D vchiq_lib: vchiq_release_message handle=00000107, header=b6a89410
01-01 00:07:32.445  1615  1621 I ilclient: cl:video_scheduler:3: callback state changed (Loaded)
01-01 00:07:32.445  1615  1621 D vchiq_lib: vchiq_release_message handle=00000107, header=b6a8d010
01-01 00:07:32.445  1615  1615 D vchiq_lib: vchiq_queue_message called service handle = 0x00000107
01-01 00:07:32.445  1615  1616 D vchiq_lib: callback(2, b6a8bc10, b6ebbf8c(107,b6c32e10), 0)
01-01 00:07:32.445  1615  1621 D vchiq_lib: vchiq_release_message handle=00000107, header=b6a8bc10
01-01 00:07:32.446  1615  1615 D vchiq_lib: vchiq_queue_message called service handle = 0x00000107
01-01 00:07:32.446  1615  1616 D vchiq_lib: callback(2, b6a8d010, b6ebbf8c(107,b6c32e10), 0)
01-01 00:07:32.446  1615  1621 D vchiq_lib: vchiq_release_message handle=00000107, header=b6a8d010
01-01 00:07:32.446  1615  1615 D vchiq_lib: vchiq_queue_message called service handle = 0x00000107
01-01 00:07:32.446  1615  1616 D vchiq_lib: callback(2, b6a8bc10, b6ebbf8c(107,b6c32e10), 0)
01-01 00:07:32.446  1615  1621 D vchiq_lib: vchiq_release_message handle=00000107, header=b6a8bc10
01-01 00:07:32.446  1615  1615 D vchiq_lib: vchiq_queue_message called service handle = 0x00000107
01-01 00:07:32.447  1615  1616 D vchiq_lib: callback(2, b6a8d010, b6ebbf8c(107,b6c32e10), 0)
01-01 00:07:32.447  1615  1621 D vchiq_lib: vchiq_release_message handle=00000107, header=b6a8d010
01-01 00:07:32.447  1615  1621 D vchiq_lib: vchiq_remove_service called service handle = 0x00000107
01-01 00:07:32.447  1615  1616 D vchiq_lib: callback(1, 0, b6ebbf8c(107,b6c32e10), 0)

**Note:** Same app works well for rpi 2 android lollipop

6by9, 2016-08-01:
Does it work on a Pi2 with N preview 2, or a Pi3 with Lollipop? Changing two variables at once makes it harder to find the issue.

As a first guess Android may have changed the default layer that the UI is rendered on, so you just can't see the video being decoded and rendered. You'd be wanting to add a OMX_SetParameter call for OMX_IndexConfigDisplayRegion, passing in a OMX_CONFIG_DISPLAYREGIONTYPE (https://github.com/raspberrypi/userland/blob/master/interface/vmcs_host/khronos/IL/OMX_Broadcom.h#L256) with set=OMX_DISPLAY_SET_LAYER, and layer=127 or other largish positive number (I can't remember the range on the layer - I doubt it is the full range of a signed 32 bit value).

Android is not an OS supported by Pi Towers, so you may get advice here, but finding a fix is probably up to others.

popcornmix, 2016-08-01:
You might want want to try this variant: https://github.com/popcornmix/userland/blob/master/host_applications/linux/apps/hello_pi/splash_video/video.c
which does render to the highest possible layer.

rockster4india, 2016-08-02:
@6by9
Actually we are not able to port either( Pi2 with N preview 2, or a Pi3 with Lollipop) . but I tried putting linux on  RPI3 and ran userland application. It worked well on that too and able to play HD video but not on android. 

Thank you for suggestion @6by9  @popcornmix . Let me try and let you know.

rockster4india, 2016-08-02:
I tried both But still I am not getting video on screen.  Is there any other **configuration** need to be changed? and  It may be a silly question . I just want to know where exactly _UI(window or surface) is created for video through userland application_. for eg: In Linux perspective , when we write or draw something framebuffer(fb) device , It will be  displayed on screen(display unit) and In android perspective , SurfaceFlinger will take care of creating surface. So where is that mechanism or api are called in userland application.
Sorry for my bad english.

popcornmix, 2016-08-02:
Can you run vcgencmd?
``vcgencmd dispmanx_list`` should show you the layers on display.

6by9, 2016-08-02:
Daft thought - has Android switched to the OpenGL driver instead of GLES?
If so then video_render is going to cause grief. Isn't DispmanX disabled in those situations? I thought I saw an exchange with Eric that set that up and failed opening the display.

HelloVideo does not render via any of the Android frameworks, nor the framebuffer. It adds a layer to the scene to be composed that SurfaceFlinger knows nothing about.

popcornmix, 2016-08-02:
@6by9 yes good point. You can't use video_render in conjunction with the the arm side graphics driver (dtoverlay=vc4-kms-v3d-overlay).

Eventually there may be a way of displaying decoded video buffer through the arm side driver, but it won't involve the current video_render.

rockster4india, 2016-08-03:
Hi,
`vcgencmd dispmanx_list` was not installed - got an error "vcgencmd: not found"

I tried dumpsys of SurfaceFlinger which also provides Layers on display.  Will that help you to predict something? 
PFA dumpsys file.
[dumpsys_SurfaceFlinger.txt](https://github.com/raspberrypi/userland/files/398632/dumpsys_SurfaceFlinger.txt)

and I am planning to go through the flow of video_render i.e from ilclient_create_component(video render) Wiil that help you to find the cause?

6by9, 2016-08-03:
It sounds like video_render is doing all the right stuff, just not getting images on the screen - the tracing you can do is unlikely to show up anything.

As above, check if you are using the OpenGL driver by looking in /boot/config.txt for a line "dtoverlay=vc4-kms-v3d-overlay". If so then video_render will not work at all as composition is controlled solely from the ARM and video_render can't insert an extra overlay.

rockster4india, 2016-08-03:
Yes It is mentioned in config.txt as
`dtoverlay=vc4-kms-v3d,cma-256 `
PFA config.txt file. 
[config.txt](https://github.com/raspberrypi/userland/files/399128/config.txt)
So What should I do now? Is there any other way to change control?(sorry if my question is wrong)

Some Addition information below which may help you 
In RPI3 at boot time I am getting log as
`loaded egl library /system/lib/egl/libGLES_mesa.so`

In RPI2 at boot time I am getting log as
```
D/libEGL  (   83): Emulator without GPU support detected. Fallback to software renderer.
D/libEGL  (   83): loaded /system/lib/egl/libGLES_android.so

```

popcornmix, 2016-08-03:
No video_render is not supposed to work with vc4-kms-v3d (which is still at an experimental stage and doesn't support hardware video decode).

The video_decode component should work, and can return pixels. You can then display these pixels through whatever means are suitable (e.g. the framebuffer or an EGL surface) if you are trying to get HW video decode working in Android.

In the future there will be a zero copy scheme to get decoded displayed with vc4-kms-v3d but the mechanism for that hasn't even been decided and it won't be happening any time soon.

rockster4india, 2016-08-04:
Thank you for your valuable comments @popcornmix @6by9

mukeshkumard, 2016-08-05:
To get pixels(YUV) from decoder after decoding I am using OMX_FillThisBuffer() in video.c under hello_video. For that I am setting portIndex for output buffer as 131 since input buffer portIndex is 130. When I call  setParameter for output port using 131, Bad PortIndex error is thrown. Is 131 a valid port index or which portIndex should I use for output port.

JamesH65, 2016-08-05:
This is NOT the place for technical help, its for bug reports. Please use
the forum.

popcornmix, 2016-08-05:
Yes, this would be a better place for support questions:
https://www.raspberrypi.org/forums/viewforum.php?f=70

closed 2016-08-05 by popcornmix