raspberrypi/userland #327: Getting decoded(yuv) data to hellovideo from decoder after decoding

mukeshkumard, 2016-08-05:
Hi,
I am running userland (hellovideo) application in Android rpi3. After fetching h264 data from file OMX_EmptyThisBuffer() is invoked to transfer the buffer to decoder. In my case after decoding the data,I am not able to see the video on screen. So to test whether the h264 data is decoded to YUV data, can I use OMX_FillThisBuffer()  in video.c to get the decoded buffer from decoder. Will the decoder give me the YUV data(back to application) through that call?

JamesH65, 2016-08-05:
This is a technical question rather than a bug report and should therefor
for be posted on the Raspberry Pi forums, not here.

mukeshkumard, 2016-08-05:
Okay ThankYou!!!

closed 2016-08-05 by mukeshkumard