raspberrypi/userland #328: Qt render performance w/ eglfs is 10x slower than x11 OpenGL ES?

fhunleth, 2016-08-05:
Using a simple Qt 5.6 app that draws rectangles and a few lines of text, I'm getting a 10x slowdown when running fullscreen with eglfs vs. x11. On a Pi Zero, the difference is roughly 350 ms/draw vs. <30 ms/draw. I can reproduce a similar slowdown on a Pi 3 as well.

I'm using a fresh `2016-05-27-raspbian-jessie.img` installation, but I've been able to reproduce the issue with other versions. To test, I installed qt5 on Raspbian, built my application, and it ran fine. Then I built Qt 5.6 with eglfs support (eglfs_brcm, specifically) on the Raspberry Pi and retested. This was 10x slower. To check my Qt 5.6 build, I did a manual build for X11 and saw good performance again.

For the eglfs build, I verified that the eglfs_brcm plugin was being run by enabling Qt qpa logging. Based on the challenges of building eglfs_brcm with Qt, I'm pretty certain that the OpenGL drivers in this project are being called. However, CPU utilization is very high.

What I'm wondering is whether there is additional setup on Raspbian that X11 does and that doesn't happen when running outside of X with eglfs. Is there a known slowdown when running OpenGL outside of X11? Is there anyway to log at the VideoCore driver level (or other level) to know whether Qt is doing something crazy that's slowing down performance or whether there's a memory configuration issue or something else? I'm using all of the Raspbian defaults right now.

popcornmix, 2016-08-05:
Are you using the experimental graphics driver? (``dtoverlay=vc4-kms-v3d``)
https://www.raspberrypi.org/blog/another-new-raspbian-release/

If you aren't I suspect you are using a software mesa implementation which will obviously be slow.

Without the experimental driver, the firmware only supports OpenGLES 2, not desktop OpenGL which I suspect you may need.
(Although I know nothing about Qt with eglfs, so I'm just guessing).

fhunleth, 2016-08-05:
Thanks for responding! I'm not using the experimental graphics driver, but that's actually pretty interesting. 

As for OpenGL vs. OpenGLES 2, I'm pretty sure that it's running against OpenGLES 2. The Qt project makes a big deal about supporting it, I got compiler errors when I didn't have the include/lib directories pointed properly to `/opt/vc/...`, and here's a snippet from the Qt `./configure` script:

```
  OpenGL / OpenVG: 
    EGL .................. yes
    OpenGL ............... yes (OpenGL ES 2.0+)
    OpenVG ............... yes-auto
  QPA backends: 
    EGLFS ................ yes
      EGLFS Raspberry Pi . yes
      EGLFS X11 .......... no
    LinuxFB .............. no
```

Other than performance, is there any way to nail down which driver (software mesa, hw accelerated, etc.) is being used behind the scenes? I'm struggling as to how to give people good data as to nail down where the problem exists. It seems like examples exist that say everything is fine individually, but yet somehow everything together is not running fine.

fhunleth, 2016-08-06:
Update: I just found out that alpha blending rectangles was really hurting performance with eglfs. It doesn't hurt it near as much with X11.

I simplified the test a lot and documented my results at https://github.com/fhunleth/qt-rectangles. Sadly, I am now getting different performance numbers from the more complicated original application, but I still think it demonstrates the slow down well.

fhunleth, 2016-08-06:
Closing. I found out that the code in question used the software renderer in every case. For the eglfs case, it would software render to an image and then draw the image using OpenGL ES. The decrease in performance is completely expected. Sorry for the noise.

closed 2016-08-06 by fhunleth