raspberrypi/userland #329: Raspiyuv padding to multiple of 16

reiner-dolp, 2016-08-11:
I am currently porting my system from raspicam v1 to raspicam v2. According to [the documentation](https://www.raspberrypi.org/documentation/raspbian/applications/camera.md):

> image buffers saved in raspistillyuv are padded to a horizontal size **divisible by 16** (so there may be unused bytes at the end of each line to make the width divisible by 16). Buffers are also padded vertically to be divisible by 16, and in the YUV mode, each plane of Y,U,V is padded in this way.

However, given a width of 3280 and height of 2464, which both are divisble by 16, the image width is padded to 3296.

I suspect that the image is **padded to multiples of 32**, as `2464 % 32 = 0`, `3280 % 32 = 16` and `3296 % 32 = 0`.

reiner-dolp, 2016-08-11:
For reference, the command used is `raspiyuv --width 3280 --height 2464 -ISO 100 -n --rgb -o -`

JamesH65, 2016-08-11:
You could be right. 6x9, you are better placed to comment on this.

6by9, 2016-08-11:
Documentation error.
YUV images are always padded to a multiple of 32 pixels on width, 16 pixels on height. RGB should only be padded to a multiple of 16 in both directions. It's all optimisations to make the 16 way vector processing side on the GPU more efficient, and that includes with YUV 4:2:0's chroma subsampling.

There is a MMAL parameter ```MMAL_PARAMETER_NO_IMAGE_PADDING``` that removes all the padding as an extra processing step. The buffer itself always has to be big enough to store the padding too, but the data gets squished up to the start of it. It's used by the V4L2 driver.
If you can cope with the padding then you'll get slightly better performance.

reiner-dolp, 2016-08-11:
Thanks, a very informative answer!

> RGB should only be padded to a multiple of 16 in both directions

As i am using the `--rgb` option, this contradicts what I observed? The image is padded to multiple of 32 pixels.

6by9, 2016-08-11:
Apologies - missed the --rgb. 
Having a closer look, the MMAL layer from the camera appears to impose the stride being a multiple of 32, so that overrides the lower level image handling library requirement of a multiple of 16.
So it is a multiple of 32 on width for both YUV and RGB formats. So many permutations available!

6by9, 2016-08-11:
Plus the RaspiStillYuv app is requesting the stride to be rounded up to a multiple of 32.
https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiStillYUV.c#L802
```format->es->video.width = VCOS_ALIGN_UP(state->width, 32);```

The firmware is just abiding by that request as it doesn't violate the firmware restrictions.

reiner-dolp, 2016-08-11:
Thanks, that resolves my issue. Since the documentation is incorrect, I am leaving this bug open.

Ruffio, 2016-08-11:
@reiner-dolp you could consider closing this issue and open one at https://github.com/raspberrypi/documentation or even better, make a PR there correcting the documentation ;-)

reiner-dolp, 2016-08-11:
@Ruffio you are right.

closed 2016-08-11 by reiner-dolp