raspberrypi/userland #337: hello_pi/hello_encode demo freezes when bitrate is configured to a low value (around 100kbps to 290kbps)

amitv87, 2016-09-28:
Made following changes in  [encode.c](https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/hello_pi/hello_encode/encode.c#L188)
~~~
bitrateType.nTargetBitrate = 250000;
~~~

and the app freezes with following output and never exits
~~~
....
Writing frame 297/300
Writing frame 298/300
Writing frame 299/300
Writing frame 300/300
Teardown.
disabling port buffers for 200 and 201...
~~~

Freezes at exactly this [line](https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/hello_pi/hello_encode/encode.c#L303)
~~~
ilclient_cleanup_components(list);
~~~

I wish to use this encoder with webrtc where the bitrates are configured randomly depending on network strength. openh264 is very slow on pi even slower than libvpx.

Please help me fix the bug,
Thanks.

luked99, 2016-10-07:
Am I right in thinking that it works fine until you come to shut everything down?

Maybe this would reveal something (but might not)

    vcdbg log msg
    vcdbg log ex
    mmal_vc_diag components
    mmal_vc_diag stats

6by9, 2016-10-07:
I believe ```sudo vcdbg log ex``` does nothing on a Pi - it's missing some of PT's magic.
Adding ```start_debug=1``` to /boot/config.txt will mean that ```sudo vcdbg log assert``` may log something useful if it really is blowing up on the GPU.

hello_encode is IL, not MMAL, so mmal_vc_diag isn't going to give anything useful.

I did notice the other day that hello_encode appears to be making the incorrect assumption that it can collect an output buffer for each input buffer.

I've just run it.
Logging shows rate control struggling and failing to achieve the target bitrate - it is producing 0 bytes (ie dropping) over 50% of the frames due to having run out of bits. Print out out->nFilledLen and you'll see that lots of them come out as 0 bytes.

It does appear there is a problem that it then doesn't clean up properly on quit if the last frame produced no output. Internal function venc_exit has logged an error of "cnt != out_cnt (162,163)", which seems to be an error due to CABAC. It then doesn't complete the exit function and leaves some state lying around :-( I'll add that to the list of things to look at, but it'll be quite a way down.

I'd suggest you try using  OMX_VIDEO_PARAM_AVCTYPE and OMX_IndexParamVideoAvc to disable CABAC (select main or baseline profiles and set bEntropyCodingCABAC=OMX_FALSE), and to see if that gives better results.
You may also be able to play with OMX_IndexParamBrcmVideoInitialQuant to set the initial quantisation values to something that allows the rate control loop to actually settle.

6by9, 2017-07-20:
@amitv87 I was trying to reproduce this issue and failing. Have you tried this recently?
Seeing as the image content is fixed it produces very repeatable image streams, so I was expecting to either get it failing every time or never.
I realise this issue has been open a while, but there haven't been many changes around the video encoder that would have effected this, and reverting those that there are doesn't seem to bring the issue back.

JamesH65, 2017-12-05:
@amitv87 Any feedback, or can we close this issue?

JamesH65, 2018-07-02:
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

amitv87, 2018-07-03:
@JamesH65  I do not work on pi anymore. So I am not sure if the problem still persists. If no else is facing this issue, I have no problem closing it.